Josverl/micropython-stubs #84: Update stubs

WerdoxDev, 2021-07-26:
it would be really nice if you guys could have updated esp8266 and esp32 (especially esp8266) stubs to the latest 1.16 micropython version.

also there are stubs like the network WLAN which dont have any intellisense

Josverl, 2021-09-07:
Hi , im just one guy :-) , but happy to add any additions and stubs that you and others bring.
the [v1.16 ESP32 stubs](https://github.com/Josverl/micropython-stubs/tree/master/stubs/micropython-esp32-1_16)) have been added 
the ESP8266 runs out of memory with v1.16 on my devices . I started on a [memory optimization](https://github.com/Josverl/micropython-stubber/tree/dev/mem_optimize) to solve this , but that makes to solution harder to as that solution  requires multiple files to be send to a board rather than a single upload. 

if there is a need , i can re-visit that. Please let me know 

for the missing signatures: 
I am aware , that info is just not included in the MicroPython firmware ( due to memory constraints) , so the stubber cannot retrieve it for you to use.
also see my comments here on Work in Progress: https://github.com/Josverl/micropython-stubs/issues/92#issuecomment-914110714

Josverl, 2021-11-13:
@all-contributors add WerdoxDev for testing

WerdoxDev, 2021-11-14:
Oh i just got an email saying that i was added as a tester in this project.
As much as i would love to help you i am a bit busy on other projects :(

So unfortunately i have to reject this request sorry!
But as soon as I'm a more free i would love to help!

Josverl, 2021-11-14:
Just to be clear: I im not asking you to do more,
Im just noting that you have helped by reporting the missing stubs. 
If you are not comfortable with that , just let me know and i'll remove the mention.

Josverl, 2021-11-14:
I took a while to optimize createstubs.py to run again on the memory constrained esp8266 devices.
to run all functionality: 
 - the list of modules needed to be stubbed has been moved to a sepearate file
 - methods have been promoted to normal functions (which apparently take less memory) 
 - createstubs has been run though minification 
 - the minified version has been cross-compiled to createstubs.mpy
 
on esp8266 this now : 
- works well on MicroPython 1.13 -- 1.17 
- on MicroPython 1.12  still run out of memory 
- does not run at all on versions 1.11 and older ( requires a different version of the cross compiler) 

Updates added  in commit : 4c74dc98c700bb9c7de62d3c9e58c19abdde1f90

closed 2021-11-14 by Josverl