Josverl/micropython-stubs #91: Jedi tips?

jdtsmith, 2021-09-03:
I'm trying to use your stubs for esp32 uPy from Jedi via python-lsp-server.  I've tried linking e.g. `/path/to/project/stubs/micropython-esp32-1_16` in to my `project` directory, but Jedi doesn't seem to find them.  Any tips on how to make the connections?  Thanks!

Josverl, 2021-09-07:
sorry not off hand,
happy to help find a solution to this but i do not get your environment, are you using vscode or an other IDE?
Assuming there is a way for Jedi+python-lsp-server to ingest custom stubs that should work the same .

so what is needed is  to find how to add that configuration, for either Jedi, or for the lsp-server or the components used by them such as pylint

looking at : [the configuration options](https://github.com/python-lsp/python-lsp-server/blob/develop/CONFIGURATION.md) I notice : 
<html>
<body>

pylsp.plugins.jedi.extra_paths | array | Define extra paths for jedi.Script. | []
-- | -- | -- | --

</body>
</html>

which appears to allow you to add additional paths. maybe that is a first step to get you config in working order.
next step would be to do similar configuration for the other components that you use ( Pylint / mypy ...) eithrher though a .toml file or per tool configuration

jdtsmith, 2021-09-07:
Thanks.  I'm using emacs with eglot.  

I actually managed to solve this myself: see [this discussion](https://github.com/joaotavora/eglot/discussions/729).  It was a bit fussy to get the correct json format mapped to the lsp server for it to configure its internal jedi with.  

BTW, I found I needed to add `micropython-1_14-frozen/esp32/RELEASE` (even though I'm on 1.16) to get things like `uasyncio.wait_for` correct.  Does that get updated much?

Josverl, 2021-09-07:
Happy to learn that you found a solution, and also happy to see that this is also useful and usable across different editors.

wrt to **uasyncio**, that has been completely re-vamped in v1.13 and is getting updates and changes in every release. 
do you have a code snippet and explanation of what is not working well for you ?
then I can look into it 

Ill keep this open as a reminder to add a bit of documentation to link to your solution for others that use a similar setup.

jdtsmith, 2021-09-07:
Perfect, thanks.  This may be LSP specific, but oddly `uasyncio` completions tie into `asyncio`.  For example, `uasyncio.wait()` does not exist.  And jedi doesn't find any relevant completions for `uasyncio.wait_for`.  

I should say it mostly works great.  All the examples in your README gif work great (except `utime` lacks `sleep` for some reason with my combo of stubs; it has `sleep_ms` and `sleep_us`).

Thanks again!

closed 2021-10-30 by Josverl