Josverl/micropython-stubs #92: class members signatures and constants not found

branquif, 2021-09-04:
Hi,

I'm trying to use pyright (I'm a big fan of VIM so would like to use pyright lsp within Neovim). I was able to make it partially work, with the exception of functions and methods signatures and constants.

the problem I'm facing can be reproduced using pyright alone (no editor involved).

I'm using the stubs for esp32 and this is the pyright configuration:

```toml
[tool.pyright]
include = ["src"]

pythonVersion = "3.5"

exclude = ["**/node_modules",
    "**/__pycache__",
    "src/experimental",
    "src/typestubs"
]

extraPaths = ["src/libs",
"all_stubs/cpython_core",
"all_stubs/micropython_cpython_core",
"all_stubs/micropython-1_16-frozen/esp32/GENERIC",
"all_stubs/micropython-esp32-1_16",
]

reportMissingImports = true
reportMissingTypeStubs = false
```

The example using is this:
```python

from machine import Pin
from dht import DHT22

sensor = DHT22(Pin(26))
led = Pin(4, Pin.IN)

```

Running pyright in the root directory I get:

```No configuration file found.

pyproject.toml file found at xx/upyright.

Loading pyproject.toml file at xx/pyproject.toml

Assuming Python platform Darwin
stubPath xx/typings is not a valid directory.
Searching for source files
Found 1 source file
xx/src/main.py
 xx/src/main.py:7:16 - error: Expected no arguments to "Pin" constructor (reportGeneralTypeIssues)
 xx/src/main.py:7:10 - error: Expected no arguments to "DHT22" constructor (reportGeneralTypeIssues)
 xx/src/main.py:8:18 - error: Cannot access member "IN" for type "Type[Pin]"
 ¬†¬†Member "IN" is unknown (reportGeneralTypeIssues)
 xx/src/main.py:8:7 - error: Expected no arguments to "Pin" constructor (reportGeneralTypeIssues)
4 errors, 0 warnings, 0 infos
Completed in 0.591sec
```

I also checked the stub code and couldn't find method signatures either (cannot find signature for Pin.init(int, int) for instance).

What could I be missing?

Josverl, 2021-09-07:
thanks for adding the detailed configuration,

A bit of background: 
the stubs that are generated on a board are rather poor, they have no parameter and type cinformation , and no parent/ subclass relations either.

in the first iterations on the stubber i was happy to get just the functions classes and methods , and get pylint to play ball with that.
But I botched up the signatures as I specified nothing , and that now leads better linters such as Pyright and Pylance astray.
so the signatures for Pin are : 
``` python
class Pin:
    def init() -> None: ...
    def irq() -> None: ...
    def off() -> None: ...
    def on() -> None: ...
    def value() -> None: ...
```
while I would like it to have much more detail 
I am working on several fixes in a parallel branch of the stubber :
 - generate non-broken signatures by using  `Any` and `...` for return and parameter values  ( just make it not very incorrect ...)
    https://github.com/Josverl/micropython-stubber/tree/dev/betterstubs 
 - generating generic stubs from  source documentation RST files 
    https://github.com/Josverl/micropython-stubber/tree/dev/libcts 
   - a bit hard; as the documentation is written for humans, not computers, and it also lacks clear info on super / subclasses
   - then (optionally) reverse merge that with the generated stubs to avoid including generic documentation that does not apply to a specific port ( seperate work ) 

the second part is work in progress, but feel free to explore as this is trying to solve exactly your issue
One of the next work-items is actually adding the class level constants as well.
I have not yet integrated the outcomes of these , but if you are interested to take it for a spin and provide some feedback on what works, and what doesn't I can push to a test branch in this repo 

below a sample of the current 'RST generated' stub fir Pin

``` python 

class Pin:
    """
    Access the pin peripheral (GPIO pin) associated with the given ``id``.  If
    additional arguments are given in the constructor then they are used to initialise
    the pin.  Any settings that are not specified will remain in their previous state.

    The arguments are:

      - ``id`` is mandatory and can be an arbitrary object.  Among possible value
        types are: int (an internal Pin identifier), str (a Pin name), and tuple
        (pair of [port, pin]).

      - ``mode`` specifies the pin mode, which can be one of:

        - ``Pin.IN`` - Pin is configured for input.  If viewed as an output the pin
          is in high-impedance state.

        - ``Pin.OUT`` - Pin is configured for (normal) output.

        - ``Pin.OPEN_DRAIN`` - Pin is configured for open-drain output. Open-drain
          output works in the following way: if the output value is set to 0 the pin
          is active at a low level; if the output value is 1 the pin is in a high-impedance
          state.  Not all ports implement this mode, or some might only on certain pins.

        - ``Pin.ALT`` - Pin is configured to perform an alternative function, which is
          port specific.  For a pin configured in such a way any other Pin methods
          (except :meth:`Pin.init`) are not applicable (calling them will lead to undefined,
          or a hardware-specific, result).  Not all ports implement this mode.

        - ``Pin.ALT_OPEN_DRAIN`` - The Same as ``Pin.ALT``, but the pin is configured as
          open-drain.  Not all ports implement this mode.

      - ``pull`` specifies if the pin has a (weak) pull resistor attached, and can be
        one of:

        - ``None`` - No pull up or down resistor.
        - ``Pin.PULL_UP`` - Pull up resistor enabled.
[... 199 lines cut ...]
#:    Selects the pin mode.
#     Pin.IN : Any
#     Pin.OUT : Any
#     Pin.OPEN_DRAIN : Any
#     Pin.ALT : Any
#     Pin.ALT_OPEN_DRAIN : Any
# .. data:: Pin.PULL_UP
#:    Selects whether there is a pull up/down resistor.  Use the value
#:    ``None`` for no pull.
#     Pin.PULL_UP : Any
#     Pin.PULL_DOWN : Any
#     Pin.PULL_HOLD : Any
# .. data:: Pin.LOW_POWER
#:    Selects the pin drive strength.
#     Pin.LOW_POWER : Any
#     Pin.MED_POWER : Any
#     Pin.HIGH_POWER : Any
# .. data:: Pin.IRQ_FALLING
#:    Selects the IRQ trigger type.
#     Pin.IRQ_FALLING : Any
#     Pin.IRQ_RISING : Any
#     Pin.IRQ_LOW_LEVEL : Any
#     Pin.IRQ_HIGH_LEVEL : Any
# .. currentmodule:: machine
```

jdtsmith, 2021-11-02:
Looks interesting. I tried out `betterstubs`.  It's behind master so missing LVGL so I couldn't stick with it, but gave it a test.  Some key things like arguments for `uasyncio.sleep()` are still missing.  Tough to pull that info out of the (non-existent) docstrings for these I bet!

I wonder if it would make more sense for python-based uPy packages to directly parse the source files themselves?  E.g. [this for `uasyncio.sleep`](https://github.com/micropython/micropython/blob/4d9e657f0ee881f4a41093ab89ec91d03613744d/extmod/uasyncio/core.py#L61).  I believe `stubgen` from `mypy` does this.  I just tried:

```
~/code/micropython/mainline_uPy/micropython/extmod                                                                                                       [master]
% stubgen -o ../../../mypy_stubs uasyncio  
```

And it did a pretty nice job producing `.pyi` files with this package, e.g.:

```python
def sleep_ms(t, sgen=...): ...
def sleep(t): ...
```

Not too many packages are possible, but perhaps a useful supplement.  In the meantime, the OP can turn off `pyright`'s missing arg warning with:

```
reportGeneralTypeIssues = false
```

Josverl, 2021-11-14:
@JD ,
FYI : the betterstubs updates are now in the main branch , and i just pushed an update to MicroPython-Stubs with most recent ESP32 and ESP8266 versions.
See :  4c74dc98c700bb9c7de62d3c9e58c19abdde1f90

I'm also using pypy.stubgen,  and initially i was very hopeful about it being to generate typeshed stubs directly from source, but I cold not find any source file at that time that had the proper annotations that it could work from.

Based on your finding I have given it another try , and now it indeed generates some useful stubs: 
I think it makes sense to add these to the set of frozen modules for the respective micropython ports, to avoid needing to add yet another 'type of stubs'
[image]

jdtsmith, 2021-11-14:
Cool thanks, I'm working with them now.  I do get some complaints about errors like uasyncio.TimeoutError not a valid Exception-derived classes.  But getting better and better...

Josverl, 2021-11-14:
wrt the `TimeoutError` issue; 

Up to 1.14 these these modules were included in the mpy manifest as frozen modules.
I think with 1.15 they have been moved to `extmods` , so it makes sense to pick them up from there and add them to frozen , at the cost of storing duplicate files.

Josverl, 2021-12-28:
It took me a while to find the root cause of the omission.
- the =re has been a change in the format of firmware manifest files.
- I borrowed some of the code that mimics the inclusion of python files as frozen modules, and this code needed to be updated to account for the change.

this is now done, and all firmwares that support uasync should now again have the uasyncio stubs 
- [esp32/GENERIC/uasyncio](https://github.com/Josverl/micropython-stubs/tree/79bd1fb99b20f30508dc2c126cdaba77db8ffe83/stubs/micropython-1_17-frozen/esp32/GENERIC)
- [stm32/GENERIC](https://github.com/Josverl/micropython-stubs/tree/79bd1fb99b20f30508dc2c126cdaba77db8ffe83/stubs/micropython-1_17-frozen/stm32/GENERIC)
- [esp8266/GENERIC](https://github.com/Josverl/micropython-stubs/tree/79bd1fb99b20f30508dc2c126cdaba77db8ffe83/stubs/micropython-1_17-frozen/esp8266/GENERIC)
-but not : 
- [esp8266/GENERIC_512K](https://github.com/Josverl/micropython-stubs/tree/79bd1fb99b20f30508dc2c126cdaba77db8ffe83/stubs/micropython-1_17-frozen/esp8266/GENERIC_512K)

closed 2021-12-28 by Josverl