Josverl/micropython-stubs #122: Question: How use stubs in pycharm (stubs without parameters warnings)?

mrkeuz, 2021-11-09:
Hi, guys, 

Really appreciate you work! Thanks! 🚀🚀🚀

Could you advise how it can set up stubs in PyCharm? I attached stubs into IDE but get warnings for all parameters, as stubs does not have arguments. I use `micropython-esp32-1_16` variant for generic `esp32` board.

Could you advise how trick PyCharm somehow for validate only function/method names?

Josverl, 2021-11-13:
The fisrst version of createsubs.py was only able to get basic info from the partial AST on MicroPython.
Also , I myself was not too famliar with the way thet Python handles / ignores type information , nor with the many different notations used to express these types in both .py and .pyi.

over time additional work has added better support form functions and class methods, to make them comply better.
Since `createstubs.py` 1.4.1 createsubs  I now try to generate prototypes that are in the format:
``` python 
def foo(*argc)->Any:
   ""
   ...
class bar():
    ""
    def foobar(self, *argc)->Any:
       ""
       ...
      
```

So the newer stubs should not throw these errors you mention.
The [documentation](https://micropython-stubs.readthedocs.io/en/latest/firmware_flat.html) (now) includes info on which stubs are generated by which version of the stubber.

you can also check out the `modules.json` of the stubs you are currently using to see which version they are.

Josverl, 2021-11-13:
@all-contributors add mrkeuz for testing

Josverl, 2021-12-29:
@mrkeuz ,

I have done my best to document how to use the current stubs in PyCharm,
but as I am not a regular pyCharm user I may have misunderstood, so it would be good if you can do a quick sanity check of this page :  https://micropython-stubs.readthedocs.io/en/latest/24_pycharm.html

closed 2022-01-10 by Josverl