Josverl/micropython-stubs #123: [Feature] Add stubs to pypi

mrkeuz, 2021-11-09:
According [PEP-0561](https://www.python.org/dev/peps/pep-0561/), stubs can be distributed via pip.

It would be good to publish all packages to pip for easy installation. Seems as you have pipeline for generating stubs it is easily to do. Just need to decompose files in right directory structure, generate `setup.py`/`project.toml` by template and upload it to pip for all board versions.

Sincerely to all your team! 🚀🚀🚀

Josverl, 2021-11-13:
Thanks for the suggestion and the 3 rockets , 
I have been considering a simpler form of distribution than just `git cloning`, but was not aware of PEP-561

I'll need to think on this , especially on how to handle the multiple different stubs that apply to a single version
and  also on where on a file system this would end up.

Could  you help by setting up a simple sample project  e.g. just distributes a single stub ?

mrkeuz, 2021-11-15:
I've prepared sample. `esp32-test-stubs` in pypi. Tested in PyCharm. Working as expected.
Sources here: https://github.com/mrkeuz/esp32-test-stubs. 

My vision is during build just create `pyproject.toml` by template in every board folder. Or maybe create just once. 

Naming conventions may contain next parts: 
- "micropython distribution name"
- "board"
- "board sub-revision" (optional)
-  version of package from `micropython` distribution (i.e. 17)

I.e.

- micropython-esp32-generic == 17
- micropython-esp32-C2 == 17
- micropython-esp8266-generic == 17
- pycopy-esp32-generic == 3.6.0

Maybe you propose something better. But main idea is able to make `pip3 install myboard-stubs` and start micropython develop.

Josverl, 2021-12-29:
I've used your examples to build a first version but have not yet pushed it to PyPi ( see below) 
also i'm trying to figure out how this could be used in PyCharm 

``` sh 
pip install "git+https://github.com/josverl/micropython-test-stubs.git#egg=micropython-library-test-stubs&subdirectory=stubs/micropython-v1_17-docstubs"
```

closed 2022-06-20 by Josverl