Josverl/micropython-stubs #434: Can't manage to get this to work for use on my LoPy4 (Pymakr). Can it even ?

LeVraiRoiDHyrule, 2022-02-04:
Hi,

I am trying to get Intellisense to work while I program my LoPy4 from Pycom with Expansion Board 3.1. If I understood correctly, it has a ESP32 chipset, so the default config from https://micropython-stubs.readthedocs.io/en/main/20_using.html should work.

I've correctly extracted the main archive next to my project, set up the symlink (it appears in Vscode explorer):
[image]

I've pasted the sample .vscode folder as well as the .pylintrc file into my project. Also, I've installed pylintrc when vscode prompted it.

But it still doesn't find most of the imports I'm willing to do:
[image]

it doesn't find Bluetooth in network for example. I know it exists because it works when I send this code to the LoPy4. But it's a problem because I can't know if a method exists before trying it on the board.

Is it even supposed to work with the LoPy ? I found no direct info on this. I'm a begginer so maybe it's just that I misunderstood something.
If anyone has any answer, feel free to help.

Thanks in advance for any help,
Have a great day

Josverl, 2022-02-09:
Hi, 
- yes, this config should work with your board 
- no, there is no complete no pre-build set of stubs for your board, ( as I do not have a WiPy ) 

pycom has a different firmware version based on a different branch of micropython, with quite a few additional modules.

In order to create `firmware stubs` for your device you need to run `createsubs.py` on your device.
The step-by-step is here : https://micropython-stubber.readthedocs.io/en/main/40_firmware_stubs.html#running-the-script

Using just these machine stubs would get you a basic experience, which is quite decent.

I have not (yet) spend efforts to also include frozen-stubs for pycom, but you could download [the -to-be-frozen  *.py modules](https://github.dev/pycom/pycom-micropython-sigfox/blob/a37510c092bcec00671c924accb97dcdfa2f4b5d/esp32/frozen) and add them to your `typings`  folder 

[micropy-cli](https://micropy-cli.readthedocs.io/en/latest/) can do this for you, but it is not under active maintenance, and is using a quite old version of my scripts :-(

Josverl, 2022-02-09:
Can you please share the output of `help("modules") ` on your board ?

LeVraiRoiDHyrule, 2022-02-09:
> Can you please share the output of `help("modules") ` on your board ?

```
>>> help('modules')
_OTA              _pybytes_machine_learning           math              uerrno
__main__          _pybytes_main     micropython       uhashlib
_boot             _pybytes_protocol network           uio
_coap             _pybytes_pyconfig os                ujson
_flash_control_OTA                  _pybytes_pymesh_config              pycom             umachine
_main             _terminal         queue             uos
_main_pybytes     _thread           re                uqueue
_mqtt             _urequest         select            ure
_mqtt_core        array             socket            uselect
_msg_handl        binascii          ssl               uselect
_periodical_pin   builtins          struct            usocket
_pybytes          cmath             sys               ussl
_pybytes_ca       crypto            time              ustruct
_pybytes_config   errno             ubinascii         utime
_pybytes_config_reader              framebuf          ubinascii         utimeq
_pybytes_connection                 gc                ucollections      uzlib
_pybytes_constants                  hashlib           ucrypto
_pybytes_debug    json              uctypes
_pybytes_library  machine           uerrno
Plus any modules on the filesystem
>>>
```
I ran createstubs.py and got the stubs folder, set it aside my project, created a symlink to it `mklink /d all-stubs D:\OneDrive\Polytech\Projets\board\stubs` in the root of my project, and edited .pylintrc like this:
`[MASTER]
init-hook='import sys;sys.path[1:1] = ["all-stubs/micropython-v1_20_2_r6-lopy4", "src/lib", "all-stubs/cpython_core-pycopy", "all-stubs/micropython-v1_17-frozen/esp32/GENERIC", "all-stubs/micropython-v1_17-esp32",];'
`
And settings.json like this:
```
{
    "python.languageServer": "Pylance",
    "python.autoComplete.extraPaths": [
        "all-stubs/micropython-v1_20_2_r6-lopy4",
        "src/lib",
        "all-stubs/cpython_core-pycopy",
        "all-stubs/micropython-v1_17-frozen/esp32/GENERIC",
        "all-stubs/micropython-v1_17-esp32",
    ],
    "python.linting.enabled": true,
    "python.linting.pylintEnabled": true,
    "python.jediEnabled": false,
    "python.analysis.extraPaths": [
        "all-stubs/micropython-v1_20_2_r6-lopy4",
        "src/lib",
        "all-stubs/cpython_core-pycopy",
        "all-stubs/micropython-v1_17-frozen/esp32/GENERIC",
        "all-stubs/micropython-v1_17-esp32"
    ],
}
```

It seems to be working, it correctly lists all available methods for Bluetooth() for example. But it doesn't tell what are the possible arguments, is it possible to get suggestions for this as well ?

[image]

Josverl, 2022-02-14:
Unfortunately the information regarding the arguments is not included in the MicroPython firmware.
As a workaround I use the *args, **kwargs ( essentially any param)  so that good code will not be marked as having errors.  

The only way to get that information is by parsing the documentation , and merging it with the firmware-stubs 
that can be done manually ( a lot of work) or automated ( much more scalable) 
im working on the automated way to do do this for standard MicroPython

Josverl, 2022-02-14:
I have added the additional pycom specific modules to the list  of modules to stub in the stubber **pycom branch:**  
-  https://github.com/Josverl/micropython-stubber/tree/pycom/board 

that adds:
 - detection of the pycom family .
   note: the foldername of the generated stubs should change to  `pycom-v1_20_2_r6-lopy4` 
- inclusion of the additional pycom specific modules

closed 2022-02-21 by Josverl