Josverl/micropython-stubs #580: Micropython-stubs not working together with Pymakr-vsc 2

sebi5361, 2022-06-23:
Dear @Josverl,
I am having an issue trying to get micropython-stubs working together with pymakr-vsc 2 when a remote location is defined.
Hence with **my_workspace.code-workspace** being:
```json
{
	"folders": [
		{
			"name": "LOCAL",
			"path": "."
		},
		{
			"name": "REMOTE",
			"uri": "serial://com11/"
		}
	],
	"settings": {
		"python.languageServer": "Pylance",
		"python.autoComplete.extraPaths": [
			"C:/Users/sebi/anaconda3/Lib/site-packages/micropython_esp8266_stubs-1.19.1.post2.dist-info",
		],
		"python.linting.enabled": true,
		"python.linting.pylintEnabled": false,
		"python.analysis.extraPaths": [
			"C:/Users/sebi/anaconda3/Lib/site-packages/micropython_esp8266_stubs-1.19.1.post2.dist-info"
		],
	}
}
```
...the stubs take for ages to load.
_E.g._ if I hover over the word **machine** in a local .py file I get  **loading...** rather than **(module) machine, Module: 'machine' on micropython-v1.19.1-esp8266** for more than 10 minutes.
Additionally, if I open a remote .py file, stubs will never show up...

As soon as I remove information related to the remote location:
```json
,
		{
			"name": "REMOTE",
			"uri": "serial://com11/"
		}
```
...stubs show right away.
Any idea for such an issue?

Josverl, 2022-06-25:
Thanks for reporting,
Have not seen this yet, but let me try to repro and find where the issue stem from.

Josverl, 2022-06-27:
@sebi5361 , I see a similar slowdown as you reported.

I created a simple repo to reproduce this .

I think this is caused by the rootpath of the mounted MCU.
for an esp8266 the rootpath is : `/` which pylance assumes is located on the local disk.
This in turn cause pylance to try to index all of your C:\ or root drive , which takes too long.
```
[Error - 08:38:04] (25540) Enumeration of workspace source files is taking longer than 10 seconds.
This may be because:
* You have opened your home directory or entire hard drive as a workspace
* Your workspace contains a very large number of directories and files
* Your workspace contains a symlink to a directory with many files
* Your workspace is remote, and file enumeration is slow
To reduce this time, open a workspace directory with fewer files or add a pyrightconfig.json configuration file with an "exclude" section to exclude subdirectories from your workspace. 
```
For more details, refer to https://github.com/microsoft/pyright/blob/main/docs/configuration.md.

I think I see the root popping up here : https://github.com/Josverl/stubs-esp-test/blob/e0445e9ddaee77d539b9969ea3b4deeeebd13a0e/ws_with_remote.log#L151-L177

the MCU service is the [workspace name for my ESP8266](https://github.com/Josverl/stubs-esp-test/blob/e0445e9ddaee77d539b9969ea3b4deeeebd13a0e/stubs-esp-test.code-workspace#L16) 

```
[Info  - 07:55:00] (35208) Setting pythonPath for service "MCU": "C:\Users\josverl\AppData\Local\Microsoft\WindowsApps\python3.9.exe"
[Warn  - 07:55:00] (35208) stubPath \typings is not a valid directory.
[Info  - 07:55:00] (35208) Assuming Python version 3.9
[Info  - 07:55:00] (35208) Assuming Python platform Windows
[Info  - 07:55:01] (35208) Search paths for \
[Info  - 07:55:01] (35208)   c:\Users\josverl\.vscode\extensions\ms-python.vscode-pylance-2022.6.30\dist\typeshed-fallback\stdlib
[Info  - 07:55:01] (35208)   \
[Info  - 07:55:01] (35208)   \typings
[Info  - 07:55:01] (35208)   c:\Users\josverl\.vscode\extensions\ms-python.vscode-pylance-2022.6.30\dist\typeshed-fallback\stubs\...
[Info  - 07:55:01] (35208)   c:\Users\josverl\.vscode\extensions\ms-python.vscode-pylance-2022.6.30\dist\bundled\stubs
[Info  - 07:55:01] (35208)   c:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\DLLs
[Info  - 07:55:01] (35208)   c:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\Lib
[Info  - 07:55:01] (35208)   c:\Users\josverl\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0
[Info  - 07:55:01] (35208)   c:\Users\josverl\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages
[Info  - 07:55:01] (35208)   c:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0
[Info  - 07:55:01] (35208)   c:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\Lib\site-packages
[Info  - 07:55:01] (35208) Adding fs watcher for library directories:
 C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\DLLs
C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib
C:\Users\josverl\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0
C:\Users\josverl\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages
C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0
C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib\site-packages
[Info  - 07:55:01] (35208) Adding fs watcher for directories:
 \
[Info  - 07:55:01] (35208) Searching for source files
[Info  - 07:55:01] (35208) Auto-excluding \$Recycle.Bin\S-1-12-1-3020494159-1235939459-3438115750-967420959\$REJ16W7.venv
```

### How to get around this ?
Need to think about this ; I see a few possible options
 1.  get pymakr to return a path to pylance that allows it to understand that its a remote . Currently it seems to overlay the PC filesystem. ( ie the root path) . this may be less of an issue for devices that use `/flash` as the root path , but there are still errors lurking if the same path can be resolved to different content/locations) 
  I think a similar overlay was seen before in pymakr2, but not quite sure where that was or how it was fixed.

 2. It should be possible to use a `pyrightconfig.json` on the mcu , or root workspace , to avoid pylance walking your c:\ or / drive

Josverl, 2022-06-27:
a few pointers to Pylance and remote workspaces :
   "Analysis is limited to opened files."

- https://github.com/microsoft/pylance-release/discussions/2696
   **This should now be partially supported for open files only**, as of the last couple of releases. Getting more than this will be challenging.
 - https://github.com/microsoft/pyrx/pull/1982

Josverl, 2022-06-28:
I can also reproduce this issue with the MemFS sample extension,
see: https://github.com/Josverl/stubs-esp-test/blob/main/repro.md#memfs-test 

So it appears that this is a **Pylance** limitation or bug

Josverl, 2022-06-29:
😬ops, now public ✅

Josverl, 2022-07-22:
There is a fix released to Pylance Preview 2022.7.41
ref: https://github.com/microsoft/pylance-release/issues/2978

sebi5361, 2022-07-22:
Indeed, it seems to work OK now using this Pylance pre-release when editing a `.py` file in the LOCAL location.
However when editing a `.py` file in the REMOTE location (`"uri": "serial://com11/"`) no stubs are shown.
Is this the expected behavior?

Josverl, 2022-07-25:
@sebi5361, 
Well, frankly I had hoped that remote locations would be treated the same as any other, however based on the comments in the PyLance/PyRight issue, remote locations are currently _not_ supported.

> we should disable all fs access for virtual workspace. we can't use vscode file provider api or new LSP file system api proposal for now. that require async file access which we currently doesn't support.

I suggest starting a [new discussion in the PyLance rep](https://github.com/microsoft/pylance-release/discussions)
to see if we can rally support for remote workspaces

closed 2022-07-25 by Josverl