Josverl/micropython-stubs #700: Trying to use this for RPI Pico W

antoinegomez, 2022-11-21:
I can see the definition on the line here:

https://github.com/Josverl/micropython-stubs/blob/983d2f351e27c3270cba49c96ea59de7e18e79b9/stubs/micropython-v1_19_1-rp2-merged/machine.py#L783

Not matching the lib for RP2040.

Here is the error I get:

[image]

Can you point me towards the right resources so I can build the stubs for it?
I am quite noob and not a versed python developer.

Thank you for your time and help.

brettnak, 2022-11-22:
Just to add some to this issue:

I've just generated stubs from my pico-w, and the definition that is generated is this:

```python
class Timer:
    ''
    ONE_SHOT = 0 # type: int
    PERIODIC = 1 # type: int
    def deinit(self, *args) -> Any:
        ...

    def init(self, *args) -> Any:
        ...
```

If you generate them, that should work for you as well, but for some reason when generating stubs micropython-stubber seems to be generating all function arguments as just `*args` and giving no type information.

brettnak, 2022-11-23:
The machine.Timber object's docs are here: https://docs.micropython.org/en/latest/library/machine.Timer.html, which seem to indicate to me that there is no `freq` parameter.

Josverl, 2022-12-04:
The documentation on this appears to be incorrect, there is a PR on already to correct this, but I plan  to patch it in the exeptionlist if that does not get merged

Josverl, 2023-01-29:
the [documentation PR](https://github.com/micropython/micropython/pull/9725) has been merged and based on that the issue should be resolved in the next release

closed 2023-03-20 by Josverl