Josverl/micropython-stubs #707: The required syntax for Pylance in settings.json appears to require updates

lalugue, 2022-12-09:
First of all, thank you for this repository

The documentation link concerned is the following: https://micropython-stubs.readthedocs.io/en/latest/22_vscode.html#configure-vscode-pylance-to-use-the-selected-stub-folders

Currently, the `settings.json` in the linked documentation page looks like the following:

```
    "python.languageServer": "Pylance",
    "python.analysis.autoSearchPath": true,
    "python.autoComplete.extraPaths": [
        "src/lib", 
        "all-stubs/cpython_core-pycopy", 
        "all-stubs/micropyton-1_17-frozen/esp32/GENERIC", 
        "all-stubs/micropyton-1_17-esp32",
    ]
    "python.linting.enabled": true,
    "python.linting.pylintEnabled": true,
```

It looks like `python.autoComplete.extraPaths` should now be `python.analysis.extraPaths` according to Pylance's [current readme](https://github.com/microsoft/pylance-release); an engineer from Microsoft also seems to support using the new syntax ([link](https://github.com/microsoft/pylance-release/issues/29#issuecomment-655303507))

It also looks like `python.analysis.autoSearchPath` should now be `python.analysis.autoSearchPaths`, which is also based on the linked readme above

A comma also looks required before the `python.linting.enabled` setting in `settings.json`

Overall, the updated `settings.json` would look like the following:
```
{
  "python.languageServer": "Pylance",
  "python.analysis.autoSearchPaths": true,
  "python.analysis.extraPaths": [
        "src/lib", 
        "all-stubs/cpython_core-pycopy", 
        "all-stubs/micropyton-1_17-frozen/esp32/GENERIC", 
        "all-stubs/micropyton-1_17-esp32",
  ],
    "python.linting.enabled": true,
    "python.linting.pylintEnabled": true,
}
```

If the changes look alright, I can also create the pull request for this

I appreciate the extensive documentation for this repository

PS: This also looks like a part of snippets you may be looking for in https://github.com/Josverl/micropython-stubs/issues/434#issuecomment-1038896584

Josverl, 2022-12-15:
Thanks for the nice words 😃

Yes I'm more than a bit behind on the docs, thanks for pointing to where improvements are needed.
A PR would be very welcome indeed

lalugue, 2022-12-18:
Thanks again for taking a look, i think this can be closed now as the change has been merged 🚀

closed 2022-12-18 by lalugue