Josverl/micropython-stubs #717: Add __call__ for machine Pin and Signal

mgsb, 2023-03-23:
Issue #92 comments had the __call__ for Pin, but it is not in the current release. It should be just a copy/rename of the value() method. Adding this for Pin and Signal will allow using pin(1) and signal(0) calls without warnings/errors.

Thanks for this project! I use it frequently with Emacs and the pyvenv and lsp-mode packages for developing my MicroPython projects.

Josverl, 2023-03-30:
Can you detail which port and versions of the package you saw this in?

There are quite a few to track through otherwise

Jos

mgsb, 2023-03-30:
I am using micropython-esp32 and I believe it is latest (1.19.1). I am willing to work on this if you could provide some guidance. I do not understand how the various "stubs" repos work, and hence, where to make the changes.

Josverl, 2023-03-31:
Thanks for the info,
I  can confirm your report 👍🏻 

1. Micropython documentation lists [`Pin __call__`](https://docs.micropython.org/en/latest/library/machine.Pin.html#machine.Pin.__call__) 
2. The docstubs have the method [`def __call__(self, x: Optional[Any] = None) -> Any:`](https://github.com/Josverl/micropython-stubs/blob/c9faf3ac2c495524fb4d491f6c45326ff565cf2c/stubs/micropython-latest-docstubs/machine.pyi#L227-L232)

3. The [esp32 boardstubs](https://github.com/Josverl/micropython-stubs/blob/c9faf3ac2c495524fb4d491f6c45326ff565cf2c/stubs/micropython-v1_19_1-esp32/machine.pyi#L138-L157) (aka firmware stubs ) do **not** list the `__call__` method 
4. But the resulting esp32 package](https://github.com/Josverl/micropython-stubs/blob/c9faf3ac2c495524fb4d491f6c45326ff565cf2c/publish/micropython-v1_19_1-esp32-stubs/machine.pyi#L750) does not have the `__call__` method.

As the process to generate the stubs is mostly automated - a manual change would get overwritten as soon as the next generation cycle runs.
( I just found that I never put the below diagram into the documentation , so thanks for helping me remember that ) 

I think there are two options to resolve this 

1. Change createstubs.py so that is does include / not skip reporting the `__call__` method. Currently it [skips/ignores all _underscored methods](https://github.com/Josverl/micropython-stubber/blob/935bd413ca85e2e3495fe4f01f234ac610d23792/src/stubber/board/createstubs.py#LL111-L112C97). 
2. Add an override option in the docstub/boardstub merge process to (forcefully) insert some classes/methods from the documentation that do not exist in the board stubs.

[![](https://mermaid.ink/img/pako:eNqFVU2PmzAQ_SsWUpR2tTSXntCqhyzKqlKjRkurVQU5GDyAu2AjY5qm2fz3-guSEJRNDjYz782MZ8bjg5dxAl7gzWYHRBmVATqgeV7xXVZiIefmM-vEH1DbedrVzRwd9X82S1ghcFOiH8uEJQyp32yGnkCiXPB_4ETrzS_k-1_QW46DHPsVbSVquJCLlGNBFjVmNIdWfmr2b-hpFVnSt69LTVLfelk9W-ngIxOAJaCcinqHBaBWdmnr3D3-NO4eDROtXsYkwrMzePjdAZ9Cu7plDaKAgVrrL5TvEGZkMODCWb2cCHYXnWVj2dGK9MhnozciF2w0ljxyAWPZBmev2BmPTqFb0w8m6OUpPVaspT1vCGbTpSr9pRMYUxZo5c6bAxnFfkPtmTauMMrTwwVlMB7JfUVZgYaqHw45TnXJCyrLLkVrmgm-2cuSs-MxCAIBDR-qfQOsWiYdEXTVbjBQqEo0opjT3vJiarpo7LlGZJ0HzW01V5a2ibXQ4JrXoj90FNsubzsqM9zChYe7uzvrZXtO6nsjXjj7Oa8ICF8KALQyF8n290LTLjtvmuNuhT9i9S2nSWOO6V4yzQgnGSrF03DdwVMELZ9mXHR5Tx0y6OSLi6SFy_iDxREscapxv1vOSPpx27ekGgSuGLWuQVbSRsu2LsoovtLaAWECU9No20eZghhs6jTFsVku9VqrplUcFyB9O_5sqa9RoUP1c-QaYq5wHJvlWuuuXhy7zVSgWYXbNoQc6R5Wc7KqgooWpUyrDu5bKfgr-DtKZBl8bv6OKM7UGatQ9WMTqNZisj1m7xlVlbsJTpgaIV2jqgnevafmbY0pUY_SQdtJPFlCDYkXqC2BHHeVTLyEHRUUd5JHe5Z5gRTqbJ61EVKsnqa6FwKhkou1fefMc3f8D99bRJU?type=png)](https://mermaid.live/edit#pako:eNqFVU2PmzAQ_SsWUpR2tTSXntCqhyzKqlKjRkurVQU5GDyAu2AjY5qm2fz3-guSEJRNDjYz782MZ8bjg5dxAl7gzWYHRBmVATqgeV7xXVZiIefmM-vEH1DbedrVzRwd9X82S1ghcFOiH8uEJQyp32yGnkCiXPB_4ETrzS_k-1_QW46DHPsVbSVquJCLlGNBFjVmNIdWfmr2b-hpFVnSt69LTVLfelk9W-ngIxOAJaCcinqHBaBWdmnr3D3-NO4eDROtXsYkwrMzePjdAZ9Cu7plDaKAgVrrL5TvEGZkMODCWb2cCHYXnWVj2dGK9MhnozciF2w0ljxyAWPZBmev2BmPTqFb0w8m6OUpPVaspT1vCGbTpSr9pRMYUxZo5c6bAxnFfkPtmTauMMrTwwVlMB7JfUVZgYaqHw45TnXJCyrLLkVrmgm-2cuSs-MxCAIBDR-qfQOsWiYdEXTVbjBQqEo0opjT3vJiarpo7LlGZJ0HzW01V5a2ibXQ4JrXoj90FNsubzsqM9zChYe7uzvrZXtO6nsjXjj7Oa8ICF8KALQyF8n290LTLjtvmuNuhT9i9S2nSWOO6V4yzQgnGSrF03DdwVMELZ9mXHR5Tx0y6OSLi6SFy_iDxREscapxv1vOSPpx27ekGgSuGLWuQVbSRsu2LsoovtLaAWECU9No20eZghhs6jTFsVku9VqrplUcFyB9O_5sqa9RoUP1c-QaYq5wHJvlWuuuXhy7zVSgWYXbNoQc6R5Wc7KqgooWpUyrDu5bKfgr-DtKZBl8bv6OKM7UGatQ9WMTqNZisj1m7xlVlbsJTpgaIV2jqgnevafmbY0pUY_SQdtJPFlCDYkXqC2BHHeVTLyEHRUUd5JHe5Z5gRTqbJ61EVKsnqa6FwKhkou1fefMc3f8D99bRJU)

Josverl, 2023-04-01:
interestingly at runtime the Pin object does not report that it is callable on esp32 ( but indeed it is) 
So unfortunately **Option 1)** will not help solve this (would have been an easy 2-line fix) 

```
('OPEN_DRAIN', '7', "<class 'int'>", 7, 1)
('IRQ_FALLING', '2', "<class 'int'>", 2, 1)
('IRQ_RISING', '1', "<class 'int'>", 1, 1)
('PULL_UP', '2', "<class 'int'>", 2, 1)
('OUT', '3', "<class 'int'>", 3, 1)
('PULL_DOWN', '1', "<class 'int'>", 1, 1)
('WAKE_HIGH', '5', "<class 'int'>", 5, 1)
('DRIVE_0', '0', "<class 'int'>", 0, 1)
('IN', '1', "<class 'int'>", 1, 1)
('WAKE_LOW', '4', "<class 'int'>", 4, 1)
('DRIVE_3', '3', "<class 'int'>", 3, 1)
('DRIVE_1', '1', "<class 'int'>", 1, 1)
('DRIVE_2', '2', "<class 'int'>", 2, 1)
('irq', '<function>', "<class 'function'>", <function>, 2)
('off', '<function>', "<class 'function'>", <function>, 2)
('on', '<function>', "<class 'function'>", <function>, 2)
('init', '<function>', "<class 'function'>", <function>, 2)
('value', '<function>', "<class 'function'>", <function>, 2)
>>> dir(machine.Pin)
['__class__', '__name__', 'value', '__bases__', '__dict__', 'DRIVE_0', 'DRIVE_1', 'DRIVE_2', 'DRIVE_3', 'IN', 'IRQ_FALLING', 'IRQ_RISING', 'OPEN_DRAIN', 'OUT', 'PULL_DOWN', 'PULL_UP', 'WAKE_HIGH', 'WAKE_LOW', 'init', 'irq', 'off', 'on']
>>> p5 = machine.Pin(5, machine.Pin.OUT)
>>> p5.value(1)
>>> p5(0)
```

Josverl, 2023-04-03:
@mgsb 

I have automated the addition  of the __call__ method 
and I have added Pin as the baseclass for Signal , so that should inherit that automatically Similar as Switch (although calling Switch makes no sense) 

Can you please test to see if this works for you ?
you can install the updated stubs from git : 
  
`pip install -U "git+https://github.com/Josverl/micropython-stubs/@add_callable#egg=micropython-esp32-stubs&subdirectory=publish/micropython-v1_19_1-esp32-stubs" --no-user --target typings`

mgsb, 2023-04-03:
I installed and see the Pin __call__ now in the .pyi and LSP sessions, but it does not show up for Signal. I do not see Signal as a subclass of Pin the .pyi.

Josverl, 2023-04-03:
I found that I/the code did not yet merge the baseclass from the documentation,
I fixed that and added some tests.

there are a few other class/baseclass pairs that need some more work , but for Signal it now should work.
[image]

That should be fixed now, can you give it another try ?

mgsb, 2023-04-03:
Thanks - both Pin and Signal `__call__` work now in LSP.

closed 2023-04-05 by Josverl