Josverl/micropython-stubs #719: Missing machine.PWM

andrewleech, 2023-05-19:
In particular I'm using the stm32 stubs and notice they're missing `machine.PWM` 
https://github.com/Josverl/micropython-stubs/blob/main/stubs/micropython-latest-stm32-merged/machine.pyi

ref: https://docs.micropython.org/en/latest/library/machine.PWM.html

andrewleech, 2023-05-30:
It took me _way_ too long to realise that stm32 port doesn't actually implement or provide `machine.PWM`. So yeah, stubs are correct!

closed 2023-05-30 by andrewleech

Josverl, 2023-06-01:
Hi Andrew, 
Sorry for not noticing earlier, but that is exactly what I wanted to solve with the stubs.

But I've been in the same place of confusion over what works where and how.

Happy to hear that this time it worked out