Josverl/micropython-stubs #720: Class io.StringIO has abstract attributes.

andrewleech, 2023-05-30:
I've just updated to `micropython_stm32_stubs-1.20.0.post1` and getting some new mypy errors flagged in `io.pyi`:
```
tools/typings/io.pyi:22: error: Class io.StringIO has abstract attributes "__enter__", "__exit__", "__iter__", "__next__", "fileno", "isatty", "readable", "readlines", "seekable", "truncate", "writable", "writelines"  [misc]
tools/typings/io.pyi:22: note: If it is meant to be abstract, add 'abc.ABCMeta' as an explicit metaclass
tools/typings/io.pyi:34: error: Class io.BytesIO has abstract attributes "__enter__", "__exit__", "__iter__", "__next__", "fileno", "isatty", "readable", "readlines", "seekable", "truncate", "writable", "writelines"  [misc]
tools/typings/io.pyi:34: note: If it is meant to be abstract, add 'abc.ABCMeta' as an explicit metaclass
```
Both StringIO and BytesIO have `typing.IO` as their base which have these abstract functions defined, but not expected to be present in StringIO and BytesIO.

I know mypy usage isn't officially supported, but thought I'd share this as I didn't see it in the previous version.

Josverl, 2023-06-01:
Thanks for the report, I've tried to test with mypy before, but that reported too many errors then.

If you can share your mypy setup / config then I'm happy to look into this, and add it to the test suite.

andrewleech, 2023-06-05:
I'm using `mypy` on just my firmware src code via [pre-commit](https://pre-commit.com/)

My `.pre-commit-config.yaml` looks similar to:
```
files: "^src/firmware"

repos:
  - repo: https://github.com/ambv/black
    rev: 22.3.0
    hooks:
      - id: black
        args: [
            "--fast", # Match micropython black configuration
            "--line-length=99", # Match micropython black configuration
          ]

  - repo: https://github.com/charliermarsh/ruff-pre-commit
    # Ruff version.
    rev: "v0.0.261"
    hooks:
      - id: ruff
        args: ["--fix"]
        additional_dependencies:
          ["micropython-stm32-stubs"]

  - repo: https://github.com/pre-commit/mirrors-mypy
    rev: "v1.2.0"
    hooks:
      - id: mypy
        additional_dependencies:
          ["micropython-stm32-stubs"]
```

and then tool / mypy settings are in `pyproject.toml`
```
[tool.mypy]

mypy_path = '''
tools/typings:
src/libs/microdot/microdot-asyncio:
src/libs/microdot/src:
'''

# Allow use of own stubs library in tools/typings
# https://micropython-stubs.readthedocs.io/en/main/28_mypy.html

use_builtins_fixtures = true
no_site_packages = true

# Don't check imported (library/stub) modules.
follow_imports = "skip"
follow_imports_for_stubs = false

# Note: pre-commit sends all changed files explicitly to mypy,
# bypassing this exclude setting. Ensure pre-commit settings
# limit checked files appropriately.
exclude = [
    'tools/typings/',
    'src/micropython/',
    'src/libs/',
]

[[tool.mypy.overrides]]
# Ignore mypy errors in stubs
module = [
    'stdlib.os',
    'pyb',
    'machine',
    'uasyncio.*',
    'manifest',
]
ignore_errors = true

[tool.ruff]
# Enable pycodestyle (`E`) and Pyflakes (`F`) codes by default.
select = ["E", "F"]
ignore = []

# Allow autofix for all enabled rules (when `--fix`) is provided.
fixable = ["A", "B", "C", "D", "E", "F", "G", "I", "N", "Q", "S", "T", "W", "ANN", "ARG", "BLE", "COM", "DJ", "DTZ", "EM", "ERA", "EXE", "FBT", "ICN", "INP", "ISC", "NPY", "PD", "PGH", "PIE", "PL", "PT", "PTH", "PYI", "RET", "RSE", "RUF", "SIM", "SLF", "TCH", "TID", "TRY", "UP", "YTT"]
unfixable = []

builtins = ["const"]

# Exclude a variety of commonly ignored directories.
exclude = [
    ".bzr",
    ".direnv",
    ".eggs",
    ".git",
    ".hg",
    ".mypy_cache",
    ".nox",
    ".pants.d",
    ".pytype",
    ".ruff_cache",
    ".svn",
    ".tox",
    ".venv",
    "__pypackages__",
    "_build",
    "buck-out",
    "build",
    "dist",
    "node_modules",
    "venv",
    "tools/typings",
]

# Same as our configuration of Black.
line-length = 99

# Allow unused variables when underscore-prefixed.
dummy-variable-rgx = "^(_+|(_+[a-zA-Z0-9_]*[a-zA-Z0-9]+?))$"

# Assume Python 3.10.
target-version = "py310"

# Avoid automatically removing unused imports in __init__.py files.
ignore-init-module-imports = true

typing-modules = ["micropython-stm32-stubs"]

[tool.ruff.mccabe]
# Unlike Flake8, default to a complexity level of 10.
max-complexity = 10

[tool.ruff.per-file-ignores]
"manifest.py" = ["F821"]  # "Undefined name" for eg. freeze, include, etc.
```

I'm explicitely excluding a few (typing) modules in the `tool.mypy.overrides` section above as they were failing mypy, but for the most part I'm getting good support from this configuraion of `mypy` and `micropython-stm32-stubs`

Josverl, 2025-02-09:
@andrewleech,

With the v1.24.1 I have been able to make significant progress on the compatibility with mypy.

There are still some issues to sort out that I am aware of, but I would be interested to learn if you see an improvement, or what the remaining issues you see that I have not yet spotted.

https://github.com/Josverl/micropython-stubs/blob/main/docs/Known_Limitations_v1.24.1.md

And part of the plan for further improvements : https://github.com/Josverl/micropython-stubs/issues/781

Josverl, 2025-02-09:
Ill close this issue as I think the original issue has been resolved.

closed 2025-02-09 by Josverl