Josverl/micropython-stubs #728: array.array is missing __setitem__ and __getitem__

mgsb, 2023-11-13:
The array class is missing the set/get methods so that indexed accesses are allowed.

Josverl, 2023-11-13:
Hi Mark,

Yes , I noticed that problem as well while doing some other updates 

I have added a quality check and a fix for that in micropython-stubber: 
https://github.com/Josverl/micropython-stubber/commit/174196941c3810b119d9a1a529d68018442fa978

I plan to release a set of new stubs somewhere this week.
any specific port that is relevant to you ?

mgsb, 2023-11-13:
Thanks. I mostly use esp32-merged. I continue using stubs with pyright and lsp-mode in Emacs - great stuff!

Josverl, 2023-11-18:
@mgsb,
this seemingly simple fix took longer than expected as the cause turned out to be the changes in naming convention for the boards, in combination with the old names still being present in the files and folders in the repo.

It took me some time longer to find that, and clean it up.

as you mentioned you were using  (a copy of) the esp32-merged the below is likely simpler 
You can install the latest esp32 stubs using `pip install -U micropython-esp32-stubs --target typings --no-user`

mgsb, 2023-11-19:
Thanks for looking at this, but I don't see any change(s):

mark@sam:/tmp$ pip install -U micropython-esp32-stubs --target typings --no-user --upgrade
Collecting micropython-esp32-stubs
  Using cached micropython_esp32_stubs-1.21.0.post3-py3-none-any.whl (150 kB)
Collecting micropython-stdlib-stubs>=0.9.0
  Using cached micropython_stdlib_stubs-1.0.0-py3-none-any.whl (121 kB)
Installing collected packages: micropython-stdlib-stubs, micropython-esp32-stubs
Successfully installed micropython-esp32-stubs-1.21.0.post3 micropython-stdlib-stubs-1.0.0

In the repo (after a pull), I only see the methods in the micropython-docs-stubs/array.pyi file.

Josverl, 2023-11-20:
I'll re-open this until we got it figured out. 

The way this is implemented in the stubs is that `array` is a subclass of `List`
The published version has :
`class array(List):`
https://github.com/Josverl/micropython-stubs/blob/5e882b7be58ca15b5c72f2aac241c089352cb736/publish/micropython-latest-esp32-stubs/array.pyi#L15C1-L15C19

You should see the same  in the installed version.

I verify the correctness using this code snippet : 
https://github.com/Josverl/micropython-stubs/blob/5e882b7be58ca15b5c72f2aac241c089352cb736/tests/quality_tests/feat_stdlib/check_array.py#L1-L8

If you have some code that fail , then we can 
- add it to the test to show the error 
- improve the stubs so that the error is resolved/

mgsb, 2023-11-21:
Sorry - I see it now. Not sure what I did wrong the other day - possibly did not restart the LSP server? I am confused why the stubs/ directory has a different version of array.pyi than the publish directory. I guess a rule is to not look in the stubs/ directory as there is post-processing happening into the publish/ directory?
I appreciate your support - thanks!

Josverl, 2023-11-21:
Yes, the published stubs have the highest quality. 
The others are only tested to not have syntax or semantic errors. 

It's all about the order and combination of different info from different sources.
It has grown over time, so it's not obvious.

closed 2023-11-21 by Josverl