Josverl/micropython-stubs #729: micropython stdlib contains some modules that are not available on MicroPython

Josverl, 2023-12-13:
Raised on https://github.com/paulober/MicroPico/issues/170

Some of these packages / modules are used internally by stdlib - so they cannot just be removed

- [ ] enum 
- [ ]

gilesknap, 2023-12-14:
Hi @Josverl II'm trying to use the above to get the typing module.

I've done `pip install micropython-stdlib-stubs --pre  --target typings --no-user` and I see a new 'typings' folder in my MicroPython project. 

I've added an import from `typings.stdlib.typing` example code here: https://github.com/gilesknap/pico-xmas2/blob/74f0b45949fd3d51e3e83bf4818f6e5f2dbd4f79/asyn/button.py#L5-L26

I then use vscode and MicroPico extension to upload the project. When I try to run, I get the error;
```
>>> 
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
ImportError: no module named 'typings'
```

Indeed it does look as though the typings folder is not uploaded to the pico.

Josverl, 2023-12-15:
Ah, that is something different.
The typing module is not available on micropython, only on your pc as part of the stubs.

There is a PR open to add that functionality, but there is no consencus yet.

Meanwhile you can add it to your source.

I'll share a sample of a project where I have done that

gilesknap, 2023-12-15:
OK thanks, I'll try that over the weekend. 

Now I'm not sure what your std-lib stubs added for my setup. Before I added your stubs vscode was allowing me to add typing info and I'm guessing that is because it's language server is reading my default system python libraries.

Does that make sense? Perhaps this question needs to go to @paulober as I'm not sure how his vscode extension makes use of the stubs?

paulober, 2023-12-15:
The settings.json in your pico project redirects the Python language server for stubs into `.vscode/Pico-W-Stub`.

Josverl, 2023-12-15:
@gilesknap 
- just delete the entire typings folder 
- then run `pip install -U micropython-stdlib-stubs --pre  --target .vscode/Pico-W-Stub --no-user` 
to update the stdlib stubs in that location

gilesknap, 2023-12-15:
Hi @Josverl @paulober thanks for your help on this. I now have it working.

It seems the only thing I needed to do was to add the typing.py and typing_extensions.py stubs to the lib folder. That's it.

I did successfully install your stubs `1.0.1.a1` into `.vscode/Pico-W-Stub` but it does not make any difference as type checking in my vscode editor was already working anyway (not entirely sure why?).

But anyway the result is great - I get to use all the usual power of vscode and python static type checking for code that can run on the Pico.

Many thanks!

Josverl, 2023-12-26:
Thanks for confirming.
> add the typing.py and typing_extensions.py stubs to the lib folder.

I'll add that to the documentation and sample repro.

Still thinking of a way to automate this, possibly using a mip install

Josverl, 2025-02-09:
closing old / inactive issues