Josverl/micropython-stubs #747: stdlib stubs contain sys.executable also in esp32 package

paulober, 2024-01-18:
In the esp32 micropython 1.21.0.post4 contains stdlib stubs 1.1.1. The sys.pyi stub file in the esp32 package does correctly not contain a `executable` property but the stdlib folder includes a sys folder with a __init__.py than contains a `executable` property.
That causes the pylance to show sys.executable in autocompletion even though the esp32 Micropython port doesn't include a `executable` property in the (u)sys module.

Josverl, 2024-01-18:
Hi Paul,
thanks for the report, harmonizing stdlib with the different ports is a work in progress.
pyright and mypy have different orders of preference and resolution mechanisms for stdlib modules. 

In order to get mypy to work I have recently created an automated process to refreshed the stdlib from typeshed. 

I'll need to come up with an approach to patch the module attributes as well.

Josverl, 2024-01-28:
`sys.executable` and a bunch of other module level attributes have been removed.
I published an update to stdlib that should be picked up by all the stub distributions.

See: https://github.com/Josverl/micropython-stubs/releases/tag/v1.1.2-micropython-stdlib

closed 2024-01-28 by Josverl