Josverl/micropython-stubs #753: Not clear on how to properly install stubs?

VirtualWolf, 2024-04-12:
First a disclaimer, forgive me if this seems dense but I'm relatively new to this. 😅

I was looking at the [list of firmware](https://micropython-stubs.readthedocs.io/en/main/firmware_grp.html), I have an Unexpected Maker FeatherS2 ESP32 board and want to install the "frozen" stub for MicroPython 1.22.2 for that specific board. I see it exists on that page towards the bottom, and in the repository [here](https://github.com/Josverl/micropython-stubs/tree/main/stubs/micropython-v1_22_2-frozen/esp32/UM_FEATHERS2), but if I [search PyPi for just "frozen"](https://pypi.org/search/?q=frozen&o=&c=Programming+Language+%3A%3A+Python+%3A%3A+Implementation+%3A%3A+MicroPython) there's only a handful of random boards listed, none of them up to date, and no sign of the FeatherS2 at all.

Are the stubs/frozen stubs/etc. in this repository meant to be _all_ published to PyPi automatically? If not, is there some other specific method that's recommended to get them installed?

Thanks!

Josverl, 2024-04-12:
I publish a combination of the firmware, do stubs and frozen stubs to Pypi.

As I don't have the firmware stubs for your board specifically, you could install the `micropython-esp32-stubs` and download and add any extra frozen stubs.

Or you could run the firmware stubber on your board, and generate a complete package yourself.
Happy to assist as the docs on that are not yet written...

VirtualWolf, 2024-04-13:
> Or you could run the firmware stubber on your board, and generate a complete package yourself.
> Happy to assist as the docs on that are not yet written...

Oh sure, happy to help with running that!

Josverl, 2024-04-14:
OK, I'll create some steps for you to follow

VirtualWolf, 2024-05-02:
Oops, sorry for the long delay, for some reason GitHub didn't actually send me a notification of your comment!

It was looking good right up until the end when it fell over:

```
(.venv) [virtualwolf@shade:~/Downloads/stubber]$ stubber get-mcu-stubs
21:59:28 | INFO     | cli                - micropython-stubber 1.20.0
[...]
22:04:08|INFO    |runner               - All modules have been processed, Finalizing report
22:04:08|INFO    |runner               - Path: /remote/stubs/micropython-v1_22_2-esp32-ESP32_GENERIC
22:04:08|INFO    |runner               - Local directory /var/folders/44/4d8w03kx2_73d_ps6vr5smvr000101/T/board_stubber0ycjf9wc is mounted at /remote
22:04:09|INFO    |post                 - Running autoflake on: /var/folders/44/4d8w03kx2_73d_ps6vr5smvr000101/T/board_stubber0ycjf9wc/stubs/micropython-v1_22_2-esp32-ESP32_GENERIC
22:04:10|SUCCESS |mcu_stubber          - Stubs generated for esp32-ESP32_GENERIC
22:04:10|SUCCESS |mcu_stubber          - Stubs copied to repos/micropython-stubs/stubs/micropython-v1_22_2-esp32-ESP32_GENERIC
22:04:10|INFO    |mcu_stubber          - Merging stubs with docstubs : {'version': '1.22.2', 'mpy': 'v6.2', 'port': 'esp32', 'board': 'ESP32_GENERIC', 'family': 'micropython', 'build': '', 'arch': 'xtensawin', 'ver': '1.22.2', 'cpu': 'ESP32'}
22:04:21|INFO    |merge_docstubs       - checking 1 possible board candidates
22:04:21|INFO    |merge_docstubs       - Merge v1.22.2 docstubs with boardstubs to micropython-v1_22_2-esp32-ESP32_GENERIC-merged
22:04:33|INFO    |merge_docstubs       - merged 1 of 1 candidates
22:04:33|INFO    |mcu_stubber          - Building package for {'version': '1.22.2', 'mpy': 'v6.2', 'port': 'esp32', 'board': 'ESP32_GENERIC', 'family': 'micropython', 'build': '', 'arch': 'xtensawin', 'ver': '1.22.2', 'cpu': 'ESP32'}
22:04:34|INFO    |publish              - checking 1 possible board candidates
22:04:34|INFO    |stubpackage          - Build: micropython-v1_22_2-esp32-esp32_generic-stubs
22:04:34|INFO    |stubpackage          - - Update micropython-v1_22_2-esp32-esp32_generic-stubs
22:04:34|ERROR   |stubpackage          - Exception on process, [Errno 2] No such file or directory: 'poetry'
22:04:34|INFO    |stubpackage          - micropython-esp32-esp32_generic-stubs: skip - Could not build/update package
```

I did a `pip install poetry` and reran it and it worked! 🎉 I'll see about submitting a PR this weekend with those docs polished a bit. (This was all in a venv with Python 3.12.2 under macOS 14.

In terms of boards where the repository _does_ have the stubs for it, which was more of the impetus for my original question, would the process to get the typings installed for VSCode be something like:

* Clone this repository locally
* Run `pip install /path/to/this/local/repo/publish/<directory with specific board> --target ./typings --no-user`
* Update the `.vscode/settings.json` file to add the `typings` directory

?

My main confusion was that just running `pip install` mostly appears not to work because PyPi doesn't have the vast majority of the typings...

Josverl, 2025-03-17:
closing as I think this is now well documented

closed 2025-03-17 by Josverl