Josverl/micropython-stubs #755: Attempt to create a subclass of a generic class results in TypeError: '_AnyCall' object isn't subscriptable

ANogin, 2024-05-18:
I have

```
T = TypeVar('T')

class Foo(Generic[T]):
   ...

class Bar(Foo[T]):
   ...
```

And micropython complains:
```
TypeError: '_AnyCall' object isn't subscriptable
```
for the `Foo[T]`...

Josverl, 2024-05-18:
Can you add some more detail on the type checker you are using, the python version on your host and the stubs package and version you are using?

Josverl, 2024-05-20:
If I test your sample in Pyright's playground it also shows an error ( but a different one ) 

Code sample in [pyright playground](https://pyright-play.net/?code=FAMwTg9gtgBALgTwA4EsB2BzGKpImOGAFWQFMA1AQzABoYBxUtUsFAYzoBkUBnOOogFckAG1LBgRGAF5iZKmAAUAciLKAlBLYjKPHjABiECIsbNWbANpEAuuoBcwGM4B0biQGIYAZWik4ABboWAAmEKT6aBCEpAAevHBaOnowAEKUAF6KRhCWvlD%2BQZh2js4wbi6ePn6BwTDMpCH6cBAwcQkwwPmFdbLcfNbCYjZJuvrpSjl5NUUYJU6u7kA)

```python

from typing import TypeVar, Generic, List, Tuple

T = TypeVar('T')

class Foo(Generic[T]):
   ...

# Something does not exist

class Baz(Foo[Something]):
   ...

# Something needs to exist 
Something = List[Tuple]

class Bar(Foo[Something]):
   ...

```

ANogin, 2024-05-20:
Couple of things, sorry if not clear:
- I tried to outline a hist of what my actual messy code is doing, and used `Something` as a placeholder for something complicated I have defined in my actual case.
- It type checks in mypy (as it should), but fails when actually trying to run in micropython.

Josverl, 2024-05-25:
Then it may be that you need to add the round brackets , that I tend to forget as well 
So rather than `inputs = Registry[Pin]` use `inputs = Registry[Pin]()`

Or in your sample `foo = Foo[Something]()`

ANogin, 2024-05-29:
@js

> Then it may be that you need to add the round brackets , that I tend to forget as well So rather than `inputs = Registry[Pin]` use `inputs = Registry[Pin]()`
> 
> Or in your sample `foo = Foo[Something]()`

Note that I do not have any `foo` - I have a subclass definition. Again, note that this works correctly with mypy and the syntax is correct. The issue is that the hack you use to define `Generic` causes subclasses to fail in micropython.

ANogin, 2024-05-29:
E.g.
```
% cat test.py 
from typing import TypeVar, Generic, Any

T = TypeVar('T')

class Foo(Generic[T]):
   pass

class Bar(Foo[Any]):
   pass
% mypy test.py 
Success: no issues found in 1 source file
% python3 test.py 
% wget https://github.com/Josverl/micropython-stubs/raw/main/mip/typing.mpy
--2024-05-28 21:12:05--  https://github.com/Josverl/micropython-stubs/raw/main/mip/typing.mpy
Resolving github.com (github.com)... 140.82.116.3
Connecting to github.com (github.com)|140.82.116.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/Josverl/micropython-stubs/main/mip/typing.mpy [following]
--2024-05-28 21:12:06--  https://raw.githubusercontent.com/Josverl/micropython-stubs/main/mip/typing.mpy
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1669 (1.6K) [application/octet-stream]
Saving to: ‘typing.mpy’

typing.mpy                    100%[=================================================>]   1.63K  --.-KB/s    in 0s      

2024-05-28 21:12:06 (15.9 MB/s) - ‘typing.mpy’ saved [1669/1669]

% micropython
MicroPython v1.22.2 on 2024-02-20; darwin [GCC 4.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 8, in <module>
TypeError: '_AnyCall' object isn't subscriptable
```

Josverl, 2024-07-04:
@ANogin ,

I think I can solve your problem based on Andrew`s suggestion.
But the only testing in this thusfar has been on your code snippet so please let me know if this works for you.

**typings.py** 
``` py
class _AnyCall:
    def __init__(*args, **kwargs):
        pass

    def __call__(*args, **kwargs):
        pass
    def __getitem__(self, arg): # <--
        return _anyCall

_anyCall = _AnyCall()

class _SubscriptableType:
    def __getitem__(self, arg):
        return _anyCall
```

Josverl, 2024-07-05:
I have merged the updated typings.py/.mpy 
and added a number of test cases.
See: https://github.com/Josverl/micropython-stubs/tree/main/tests/quality_tests/feat_typing

closed 2024-07-05 by Josverl