Josverl/micropython-stubs #756: Stubs for the micro:bit V2 version of MicroPython?

ddribin, 2024-06-22:
Hello,

Not sure if this is possible, given the current situation, but it would be great to have stubs for the micro:bit V2 version of MicroPython:

  https://tech.microbit.org

Currently, only their online editor is main way to code for it:

  https://microbit.org/get-started/user-guide/python-editor/
  https://tech.microbit.org/software/micropython/

But ideally, I really want to use Visual Studio Code locally.

What makes this maybe a bit tricky is that I believe they have their own fork of MicroPython:

  https://github.com/microbit-foundation/micropython-microbit-v2
  https://microbit-micropython.readthedocs.io/en/v2-docs/devguide/flashfirmware.html#micro-bit-v2

They do have their own stubs for their own online editor:

  https://github.com/microbit-foundation/micropython-microbit-stubs

But I cannot figure out any good way to use these stubs in VS Code. Ideally, they could be `pip` installed just like the ones here.

Happy to help, where possible, but I'm not even sure where to get started.

Josverl, 2024-06-24:
Hi Dave,

Ive looked before at automatically generating stubs for the MicroBit v1 - and found that it lacks the ability to write files to a filesystem. Not sure if that is any different for the v2.

The microbit stubs project is indeed currently not pip installable, but that could be added with relative ease.
I've left a comment in the thread you started.

Another microbit project that has already published to PyPI, and that you could install straight away  is https://pypi.org/project/microbit-stubs/ 
As they do not include the stdlib stubs - it may be usefull to add the `micropython-stdlib-stubs` as well
 
```
pip install --target typings --no-user microbit-stubs micropython-stdlib-stubs 
```

ddribin, 2024-06-27:
Thanks for the reply! If I'm using a `venv` do I need the `--target typings --no-user` options? Seems to work without those, either way. VS Code / Pylance sees the `microbit` package just fine. However, in both cases Pylance seems use the "real" `stdlib`, not those provided by `micropython-stdlib-stubs`.

Josverl, 2024-06-27:
Either typings or .venv will work.

to get pylance/pyright to use the micropython stdlib over the included typeshed ones you need to configure pylance where to look for them.

The setting is `typeshedPaths` and can be found in the user or workspace settings.
You need to point that to the folder in which the stdlib folder can be found ( and that differes depeding on the previous typings  .venv chooise, but either will work.
( may need a restart of the Pyhon langage server / vscode to be picked up ) 
https://micropython-stubs.readthedocs.io/en/main/22_vscode.html#configure-pylance-to-read-micropython-stdlib-stubs

closed 2024-07-04 by Josverl