Josverl/micropython-stubs #761: u-naming vs non u-unaming problem

Omid888, 2024-08-13:
`import json` & `import ujson` has the same effect on microcontroller when executed. But in vscode after doing `import json` the auto completion & its hint-box says you can have the `indent` parameter for the `json.dump()` function which is wrong. vscode is referencing to python's json package not the micropython's json package. This bug wasted an hour of me, so I reported to free other folks facing this problem.

Now that u-naming is going obsolete ([source](https://github.com/micropython/micropython/releases/tag/v1.21.0)) then micropico should first search in its libraries and then go to the general python ones.

P.S: Microcontroller reports this error when `json.dump(f,d,indent=4)` function is called:
`extra keyword arguments given`

Josverl, 2024-08-13:
Thanks for the report

Ive been chipping away to improve the static type checking. For stdlib modules such as json this is tedious, as I want to   avoid needing to recreate all of typeshed again.

I'm testing the stubs against py right and mypy using snippet test.
If you can give a few code snippets that show the failures and good situations, than I can add them, and then use that to try to improve the stubs.

Omid888, 2024-08-14:
You're right. This is my configuration:
```
{
    "python.linting.enabled": true,
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "micropico.syncFolder": "",
    "micropico.openOnStart": true,
    "python.analysis.typeshedPaths": [
        ".vscode/Pico-W-Stub",
        "~/.micropico-stubs/included"
    ],
    "python.analysis.extraPaths": [
        ".vscode/Pico-W-Stub",
        "~/.micropico-stubs/included"
    ]
}
```
it's the micropico's default configuration. Is there anything I could change?

Omid888, 2024-08-14:
No, It's empty. `stdlib` & some other `.pyi` files exists in `"~/.micropico-stubs/included"` folder.

P.S: I use macOS. Is this behavior intentional?

Josverl, 2024-08-14:
Micropy includes the stubs and configuration, in slightly different locations than I use, but the concepts are the same.

Usually this works just fine with micropy, 

@paulober , any suggestions?

paulober, 2024-08-15:
The `.micropico-stubs/included` folder contains the `micropython-rp2-rpi_pico_w-stubs` package.

@Omid888 I guess I know where to problem is. 
Have you created the project a while ago?
There where some changes in the location where the subs life. Your stubs configuration still included a path to `".vscode/Pico-W-Stub",` which no longer exists. So please try removing these two lines from your `settings.json`.

The issue is, that Pylance just silently fail if they encounter a entry in these settings which does not exist and stop parsing the setting.

paulober, 2024-08-15:
@Josverl Do you know if `python.analysis.stubsPath` is a replacement for `python.analysis.typeshedPaths` as I can't find the `typeshedPaths` setting in the docs anymore but the `stubsPath` setting is now described.
https://code.visualstudio.com/docs/python/settings-reference#_python-language-server-settings

Omid888, 2024-08-16:
@paulober Your suggestion (removing those two lines) was the correct solution. Thank You.
‌
P.S 1: Isn't it weird about that when pylance can't resolve the first option it skips the next configurations?
P.S 2: Should I reinstall micropico in another virtual-env for any other possible changes? (for problems like the current issue)
P.S 3: vscode warns me that `python.analysis.stubsPath` does not exist with this error: `Unknown Configuration Setting`. I do have the latest verion of vscode.

paulober, 2024-08-16:
> But there is a little issue, when I right click on json.dump() and select go to defenition (F12), it doesn't open micropython stubs & instead opens the python's json package.

As I was told in [this issue](https://github.com/microsoft/pylance-release/issues/6268). This is by design. If you want to go to the stubs you need to open the context menu and select "Go to declaration".

> P.S 1: Isn't it weird that when pylance can't resolve the first option it skips the next configurations?

I totally agree, thats why I opened an issue at the Pylance repo: https://github.com/microsoft/pylance-release/issues/6266

> P.S 2: Should I reinstall micropico in a new virtual-env for any other possible changes? (for problems like the current issue)

No you don't need to do that. In general I try to cover these changes by automatically converting old configurations to their new equivalent. But it seems like I forgot to included some for this change or I removed the conversion to early.

> P.S 3: vscode warns that python.analysis.stubsPath does not exist with this error: Unknown Configuration Setting. The correct spelling is python.analysis.stubPath. And it's value should not be a list, but a single string.

I misspelled the setting name. And yes it's now a single string. I did not find the `typeshedPaths` anymore in the settings and this one was new, so maybe it's a replacement and they still support the `typeshedPaths` one for some time.

Omid888, 2024-08-17:
@paulober 
> This is by design

It's okay. but could you please add this to the documentations (e.g the micropico github `readme.md` file)? It's not something people could guess, or do trial & error method to find. And it's very important (at least to me).

 
‌
> No you don't need to do that.

Thank you for informing us & and your excellent try to make the plugin better.

‌

> I did not find the typeshedPaths anymore

Yes, it's gone and the stub's documentions can be updated to produce less incompatibilities in future, dear @Josverl.

Josverl, 2024-08-17:
Yrs, I just did an overhaul of the docs, and I'll update this as well.
[This](https://micropython-stubs.readthedocs.io/en/main/22_vscode.html#relevant-settings) is what I have now 
I'll drop a note here after I update from this conversation so you can sanity-check.

I may as well add a page with a sample of the micropy settings and stub locations.

Josverl, 2024-08-20:
`python.analysis.stubPath` is something different from `python.analysis.typeshedPaths`
*   **stubPath** is a list of folder that contain just regular type-stubs, but they cannot override `stdlib` modules
*   **typeshedPaths** is an optinal list of max 1 folder that that allows one to override the typeshed stubs, that folder should contain a subfolder with the name stdlib. ( one could seperate it out, I do not for simplicity) .
    This setting is [documented for pyright and pylance](https://github.com/microsoft/pyright/blob/main/docs/configuration.md) , and works just fine in the workspace settings file for vscode.
    I also see it in the UX, so I guess the VSCode team just left it out of their docs : 
    [image]

TL;DR. 
To get the correct `time` and other stdlib modules for micropython, you must specify exactly one folder in `typeshedPaths`

Josverl, 2024-08-20:
tacking on a sub discussion o the MicroPico configuration, as there are a few things I notice.

```jsonc
{
    "python.linting.enabled": true,
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "python.analysis.autoSearchPaths": true,
    "python.analysis.stubPath": "typings",
    "python.analysis.typeshedPaths": [
        "~/.micropico-stubs/included"
    ],
    "python.analysis.extraPaths": [
        "~/.micropico-stubs/included",

    ],
    // MicroPico settings
    "micropico.syncFolder": "",
    "micropico.openOnStart": true,
}
```

* The setting `"python.linting.enabled": true,` is not used anymore AFAIK.
  I think that still stems from the old pylint settings, an can be safely removed.

* The `__builtins__.pyi` local module that is used to resolve a bare const ( without import it) should be in the [top project folder, or in the `stubsFolder`](https://github.com/microsoft/pyright/blob/main/docs/builtins.md). 
  In the above config it is in neither , but fortunately is still found. 
  I would not change it , but something to keep an eye on.

Omid888, 2024-08-24:
dear @Josverl and @paulober 
Thank you both for help.

closed 2024-08-24 by Omid888

paulober, 2024-09-09:
> ```json
> "python.analysis.stubPath": "typings",
> ```

@Josverl So you think I should still add the stubPath setting to the default settings with the value `typings` or the root of the stubs where the __builtins__.pyi is?