Josverl/micropython-stubs #762: Integrate or Merge rameter and return annotations

Josverl, 2024-08-29:
There are a few other initiatives that have focus on this topic for micropython 

| Header | Header | Header |
|--------|--------|--------|
| https://github.com/hlovatt/PyBoardTypeshed| no longer maintained| MIT|
| https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/| Copy of hlovat | MIT |
| https://github.com/Gobot1234/PyBoardTypeshed | significant manual update | MIT|
| https://github.com/JetBrains/intellij-micropython | base hlovat + community updates | Apache 2.0

With the new stubber capabilities to merge @overloads it would be possible to merge the better type info into the docstubs 

1. create docstubs from .rst 
2. merge in the parameter and return annotations & Overloads  
  - will need additional glue  imports and typedefs per module) 
4. then continue the normal process
   - merge with frozen 
   - merge firmware stubs 
5. Run regular typing QA tests to determine validity

Josverl, 2024-08-29:
Hello @Gobot1234, 
your fork of the pyboardtypeshed looks quite interesting. I'd like to understand a bit more about the [changes that you committed last year](https://github.com/Gobot1234/PyBoardTypeshed/commit/1413c3a874932a6687e7874bac5a598f3ac5ab6b) 
how did you create and test these?
Was that all manual effort ?

Gobot1234, 2024-08-29:
I think it was fairly manual because the generated ones weren't super correct. They were somewhat tested on an ardiuno nano ble but I never got everything working.

Gobot1234, 2024-08-29:
Saying all this I don't even remember if these stubs are correct for micropython they were for another microcontroller version of python IIRC

Josverl, 2024-08-29:
Thanks for the quick response, would be helpful if you can recall what flavor of (Micro)python that was. The included docs sure look like micropython to me.

Gobot1234, 2024-08-29:
Ah just remembered might have been circuit python

Josverl, 2025-02-05:
Implemented in by : https://github.com/Josverl/micropython-stubs/releases/tag/v1.24.1-stdlib

closed 2025-02-05 by Josverl