Josverl/micropython-stubs #763: stm32 1.23.0 packages not on pypi

victorallume, 2024-09-04:
Hi,

I'm trying to install `micropython-stm32-stubs==1.23.0.*` as per instructions at https://micropython-stubs.readthedocs.io/en/main/packages.html#micropython-v1-23-0 but pip returns
```
ERROR: Could not find a version that satisfies the requirement micropython-stm32-stubs==1.23.0.* (from versions: 1.17.post1, 1.17.post2, 1.17.post3, 1.18.post1, 1.18.post2, 1.18.post3, 1.18.post4, 1.19.1.post1, 1.19.1.post2, 1.19.1.post3, 1.19.1.post4, 1.19.1.post5, 1.19.1.post6, 1.19.1.post7, 1.19.1.post8, 1.19.1.post9, 1.19.1.post10, 1.19.1.post11, 1.19.1.post12, 1.20.0.post1, 1.20.0.post2, 1.20.0.post3, 1.20.0.post4, 1.20.0.post5, 1.21.0.post1, 1.21.0.post2, 1.22.0.post1)
ERROR: No matching distribution found for micropython-stm32-stubs==1.23.0.*
```
Looking at https://pypi.org/project/micropython-stm32-stubs/#history the latest published to pypi is 1.22.0

Josverl, 2024-09-07:
Ok let me look into that.
I still do the publication manually to avoid accidents, but the reverse is also true, 😔

Josverl, 2024-09-08:
You were correct, the stm32 were not published due to an earlier issue with mpremote i.c.w. at PYBv11 & SDCard.
i fixed that issue and filed a PR for that , but forgot to finish the publication process.
```
Published to PyPi  micropython-stm32-stubs                             1.23.0.post1                                   publish/micropython-v1_23_0-stm32-stubs
Published to PyPi  micropython-stm32-pybv11-stubs                      1.23.0.post1                                   publish/micropython-v1_23_0-stm32-pybv11-stubs
```

The PYBV11 is the only stm32 board I have, so it does not have the network drivers.
If you need stubs for a newer or networked stm32 board you can follow the process outlines here : https://micropython-stubs.readthedocs.io/en/main/diy_stubs_files.html

if you do, please share the stubs from your board.

closed 2024-09-10 by Josverl