Josverl/micropython-stubs #767: Add support for the assembly instructions in `@asm_pio` functions

DeflateAwning, 2024-11-15:
Here's an example of what the type checking looks like currently (observe all the errors) for `@asm_pio`-decorated Programmable IO assembly code blocks:

[image]

Doing a bit of digging around, I found that the `PIOASMEmit` class seems to define these functions, but there's no way to move those member functions to imports. 

I think the best solution would be to make them global functions, probably. Or maybe there's a way to highjack the `@asm_pio` decorator so that it runs the assembly only in that function.

DeflateAwning, 2024-11-18:
The code is here: https://gist.github.com/DeflateAwning/cfc26095d25390fcd3c619176c7bf23e

I was using these stubs: https://pypi.org/project/micropython-rp2-pico-stubs/

Josverl, 2024-11-18:
I have already built it , but not (yet) included it as part of the rp2 stubs.
[image]
I plan to add this in the upcoming release , but it needed some significant restructuring 

if you want it today - you need to add it manually to the stubs 
- rename or delete ....`/rp2.pyi`
- create a `rp2` folder 
- copy in the new rp2 & PIO stubs

See for more details :
https://github.com/Josverl/PIO_ASM_typing

DeflateAwning, 2024-11-18:
That's very neat, thank you very much! Looking forward to giving it a try. 

In the final release, will we need to import anything? Or it'll work automatically?

Josverl, 2024-11-18:
You'll need the somewhat ugly activation blurb 
```py
# -----------------------------------------------
# add type hints for the rp2.PIO Instructions
try: 
    from typing_extensions import TYPE_CHECKING # type: ignore
except ImportError:
    TYPE_CHECKING = False
if TYPE_CHECKING:
    from rp2.asm_pio import *
# -----------------------------------------------
```
before the first @asm_pio . 
as there are a bunch of side effects that don't apply to pure python , I would not want to enable it by default,
and I have not found a way to have the type checkers enable it by default.

there is a cpython decorator `@no_type_check` , but that is not (yet) part of micropython , 
but can be loaded.
That suppresses errors, but offers no type-hints.
https://micropython-stubs.readthedocs.io/en/main/typing_mpy.html#using-the-no-type-check-decorator

DeflateAwning, 2024-11-19:
This is perfect as-is! Thanks a lot!

Josverl, 2025-02-09:
This is now included in the 1.24.1 stubs for the rp2 port.
I have also added the above instructions on how to enable this

closed 2025-02-09 by Josverl

DeflateAwning, 2025-02-11:
Awesome, thanks