Josverl/micropython-stubs #768: Verify stub signatures for passing either an `int` or a `Pin` instance across all ports.

Josverl, 2024-11-20:
Actually I think since the two linked PRs merged it's possible to pass either an `int` or a `Pin` instance across all ports.

Definitely the case described in the original report works now, it's possible to do either `machine.ADC(int)` or `machine.ADC(machine.Pin(int))` and the former will degrade into the latter.

If someone runs across a machine API which requires a Pin object and doesn't accept an int, suggest they open a new issue and we can change it to use the existing helper functions internally.

_Originally posted by @projectgus in https://github.com/micropython/micropython/issues/4379#issuecomment-2487247935_

projectgus, 2024-11-20:
@Josverl Are you planning to check this? I'm not sure it's worth tracking in an issue otherwise unless someone plans to do it, AFAIK it's already been covered by the PRs linked to the original issue.

Josverl, 2024-11-20:
I plan to check if the stub signatures match this behavior, especially as I'm currently working to integrate method parameter typering and overloads, and I think that something as essential as Pins should be well represented in the type stubs.

Therefore I'm tracking it as a todo in the MicoPython-stubs repo.

projectgus, 2024-11-21:
Sounds great!

projectgus, 2024-11-21:
Oh, I'd also missed this is the micropython-stubs repo. My mistake. :facepalm: 

(Because it mentioned me I got an email notification for it that showed up with all my other MP related notifications, and I totally missed that we're no longer in the micropython org. No worries.)

Josverl, 2025-02-09:
part of the published v1.24.1 stubs

closed 2025-02-09 by Josverl