Josverl/micropython-stubs #770: sys.print_exception is not known 1.23

maxi07, 2024-12-27:
When calling

```python
import sys
sys.print_exception(e)
```
I get the error message, that `print_exception` is not known - but it works fine when running on Micropython 1.23. So it seems like the stub is missing.

Josverl, 2024-12-27:
Thanks for the report,

What port and board are you using, and which type checker is reporting the issue ?

If you are using mypy then it may be beyond its capabilities as it does not allow an alternative stdlib to be used

maxi07, 2024-12-27:
I am using a raspberry pico w (1st gen) with vscode and pylance. I got the micropico extension for vscode.

jeff-phil, 2025-01-15:
Same issue with esp8266 board, and pyright and basepyright checkers.  Thanks!

Josverl, 2025-01-19:
sorry for the delay in responding,

Have you configured the correct path to `typeshedPaths` in vscode or pyproject.toml ?

https://micropython-stubs.readthedocs.io/en/main/22_vscode.html

maxi07, 2025-01-19:
this is the content of my settings.json:

```json
{
    "python.analysis.typeshedPaths": [
        "~/.micropico-stubs/included"
    ],
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.extraPaths": [
        "~/.micropico-stubs/included"
    ],
    "micropico.syncFolder": "",
    "micropico.openOnStart": true,
    "python.testing.pytestArgs": [
        "tests"
    ],
    "python.testing.unittestEnabled": false,
    "python.testing.pytestEnabled": true,
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "python.linting.enabled": true,
    "python.terminal.activateEnvironment": true
}
```

when I manually search for the file, it is located at `~/.micropico-stubs/included/sys.pyi` - here I can actually see the print_exception function.

But if I actually use

```python
import sys
sys.print_exception()
```
and right click the sys -> goto definition, I would end up in `~/.micropico-stubs/included/stdlib/sys/__init__.py`, where the print_exception is missing.

Header of `~/.micropico-stubs/included/sys.pyi`:
```python
"""
System specific functions.

MicroPython module: https://docs.micropython.org/en/v1.23.0/library/sys.html

CPython module: :mod:`python:sys` https://docs.python.org/3/library/sys.html .

---
Module: 'sys' on micropython-v1.23.0-rp2-RPI_PICO_W
"""

# MCU: {'build': '', 'ver': '1.23.0', 'version': '1.23.0', 'port': 'rp2', 'board': 'RPI_PICO_W', 'mpy': 'v6.3', 'family': 'micropython', 'cpu': 'RP2040', 'arch': 'armv6m'}
# Stubber: v1.23.0
```

Header of `~/.micropico-stubs/included/stdlib/sys/__init__.py`
```python
"""
System specific functions.

MicroPython module: https://docs.micropython.org/en/v1.23.0/library/sys.html

CPython module: :mod:`python:sys` https://docs.python.org/3/library/sys.html .
"""
```

maxi07, 2025-01-19:
I should have added the information that I am working with the [paulober/micropico](https://github.com/paulober/MicroPico) extension, that did the setup for me

maxi07, 2025-01-20:
See my previous comment, while the sys.pyi file actually does exist, it would point me to `~/.micropico-stubs/included/stdlib/sys/__init__.py`

Josverl, 2025-01-20:
one one hand it is clear that it exists - on the other hand it can't be found.
Looking for possible cause : 
- one path uses a relative path structure that is supported by just about any OS on the planet
- the other uses a `~`  , that even on OO+Ses that support that , requires a different handling of paths 

@paulober , I think you have experimented with some different settings for the `typeshedPaths`. 
Was there any clear outcome ?

paulober, 2025-01-23:
I was able to repo the same issue.

I don't remember why we need the `typeshedPaths` setting which isn't event documented public in the first place.
I was able to solve the issue by adding following setting:
```json
"python.analysis.stubPath": "~/.micropico-stubs/included"
```
Removing the `typeshedPaths` setting or changing it to `..../stdlib` (for example) also solves this specific issue so I would assume that the `typeshedPaths` overwrite the `extraPaths` and because `typeshedPaths` expects every module to have it's own sub directory `stdlib/sys` is favored over `sys.pyi` (which is only detect by `extraPaths` which seems to be lower ranked than the other setting).

Josverl, 2025-01-24:
Thanks Paul,

The tools behaviour and settings change over time.
I do test for correct operations, but not using the extrapath settings.

Maybe I should add that to the test suite, if that is indeed a supported config by pyright.

I'm currently redesigning the stdlib stubs, so would worth checking

Josverl, 2025-01-28:
@paulober , @maxi07 

I think I have quite decent stubs for v1.24.1 , 95%+ done 
I'd appreciate your feedback on them 

The below is for pico_w stubs to ~/.micropico-stubs/included 
there are also pico and pico2 stubs if you want to check them out 

```
pip install "git+https://github.com/Josverl/micropython-stubs@reference/rp2#subdirectory=publish/micropython-stdlib-stubs" --target ~/.micropico-stubs/included
pip install "git+https://github.com/Josverl/micropython-stubs@reference/rp2#subdirectory=publish/publish/micropython-v1_23_0-rp2-rpi_pico_w-stubs" --target ~/.micropico-stubs/included
```

paulober, 2025-02-05:
> I think I have quite decent stubs for v1.24.1 , 95%+ done I'd appreciate your feedback on them
> 

Thanks for your work. I just tested them with Pico 2 and Pico W and they work great. I'll reach back if I see any issues.

Josverl, 2025-02-05:
Fixed by : https://github.com/Josverl/micropython-stubs/releases/tag/v1.24.1-stdlib

`pip install micropython-stdlib-stubs==1.24.1`

closed 2025-02-06 by Josverl