Josverl/micropython-stubs #771: time.mktime() does expect one positional argument

maxi07, 2024-12-29:
The function `time.mktime` actually does expect one positional argument: 
```python
>>> import time
>>> time.localtime()
(2024, 12, 29, 17, 33, 32, 6, 364)
>>> time.mktime()
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "<string>", line 1, in <module>
TypeError: function takes 1 positional arguments but 0 were given
Exception occured
```

but the stub expects zero arguments.

Josverl, 2024-12-30:
Thanks for the report, ill add a test for this one as well.

- time.mktime is [correctly typed in the reference-stubs](https://github.com/Josverl/micropython-stubs/blob/14b0abaa1feb48ee36057b35f45ba309f6786613/micropython-reference/time/__init__.pyi#L93)
- but currently the published stubs are indeed not correct, as these are based on the [incorrect documentation](https://docs.micropython.org/en/latest/library/time.html#time.mktime) only

Im working on fixing this and many other mismatches and emissions, and keeping the docs easy to maintain and readable by humans and typecheckers

Josverl, 2025-01-29:
Note that this issue should also be fixed in the new stubs mentioned in  https://github.com/Josverl/micropython-stubs/issues/770#issuecomment-2620230022

Josverl, 2025-02-05:
Fixed by : https://github.com/Josverl/micropython-stubs/releases/tag/v1.24.1-stdlib

`pip install micropython-stdlib-stubs==1.24.1`

closed 2025-02-05 by Josverl