Josverl/micropython-stubs #774: Built-in libs not recognized ny Mypy

Sylvain-BROCAS, 2025-01-09:
Hello,
Just a small question about micropython stuff.

I tried to use [Micropico](https://github.com/paulober/MicroPico) to develop a micropython project in VsCode. this extension uses your stubs and recomands to post here for questions linked to linting problems. So here is my problem, I hope somebody will have a solution.

 I use Mypy to check types and imports in vscode. I setup a micropython project with Micropico and wrote a simple test file to import machine, some personnal libs and a few more lines to test autocompletion.
It works great (Thank you ^^) but micropython built-in librairies seem to be unredcognized by mypy that underlines the imports in red (see the linked picture).
[image]

Maybe it's more a problem with mypy settings, but i hope I,LL find some help here.
Sylvain

Josverl, 2025-01-12:
In vscode you can use both pylance and mypy.
Pylance has my personal preference as its faster and can handle custom stdlab and builtins, where mypy cannot,
Mypy can work with some limitations and restrictions as well, and ive seen it catch some things that PyLance did not.

to configure please see:
https://micropython-stubs.readthedocs.io/en/main/28_mypy.html

MicroPico stores its typing folders in a different location, please adjust the configuration accordingly.

As mypy cannot deal woth consom stdlib , specifically `os` and `sys`, you will need to delete these from the typing folder to get mypy to work at all. this also means that you will get incorrect typehints for these modules.

```
del typings/os.pyi
del typings/sys.pyi
```

I suggest testing by running mypy from the commandline intitially.

Sylvain-BROCAS, 2025-01-13:
Ok, I will choose easy going. i'll stay with pylance.

Thank you very much for your answer

closed 2025-01-13 by Sylvain-BROCAS