Josverl/micropython-stubs #778: Asyncio wait and cancel

bixb922, 2025-01-19:
Hello,
this really great work, extremely useful! Must have been a lot of work putting all this together! I could catch several potential problems, that alone justified using the stubs. A big thank you.

I am using VSCode with Pylint. I installed the stubs for Micropython 1.23 for ESP32-S3 generic. 

With this test code, I get the indicated messages:
```py
import asyncio
x=asyncio.Event()
async def test():
    await x.wait()
    z = asyncio.create_task( test() )
    z.cancel()

# Error detected on line 4: await x.wait()
# Object of type "Generator[Unknown, None, None]" is not callable
#  Attribute "__call__" is unknown Pylance(reportCallIssue)
# (variable) wait: Generator[Unknown, None, None]

# Error detected on line 6: z.cancel()
# Cannot access attribute "cancel" for class "Task"
#  Attribute "cancel" is unknown Pylance reportAttributeAccessIssue
# (function) cancel: Unknown
``` 
Please could you help me and indicate what I'm doing wrong?

Josverl, 2025-01-20:
Its not you, the current stubs are not correct.
Up till now, I have used a combination of stop generated from the. python files. and stops generated on an MCU board. and there was a few more than a few mismatches between the two that mingled up the stubs. causing several errors likely such as the one that you've seen.

for v 1.24.x I am in process of making significant changes. One of the changes that I'm making in the current release is that I've. really worked around all the issues that I am aware of including finally tackling the errors that I knew that were lingering in Async if you're interested in validating that, I expect to make a release available for for testing in the next couple of weeks. So please let me know the port that you're working on and then I can send you a link.

Josverl, 2025-01-25:
I did more testing, 
The stubs I linked to have problems as well.

I have cobbled together a set of asyncio stubs that pass QA testing, but I'll need to find a way to integrate them in micropython-stdlib-stubs in a maintainable manner.

bixb922, 2025-01-27:
Thank you for the prompt reply.  I thought perhaps something was not well installed.  I'll gladly test as soon as new versions are available. About the port: I am using a generic ESP32-S3.

Josverl, 2025-01-28:
You can install directly from github - its a bit slow and you will need to specify the destination 
I suggest to install to a local typings folder - and then copy the contents of that to the same location as MicroPico uses  

Updated std-lib  * esp_s3
```
pip install "git+https://github.com/Josverl/micropython-stubs@reference/rp2#subdirectory=publish/micropython-stdlib-stubs" --target typings
pip install "git+https://github.com/Josverl/micropython-stubs@reference/rp2#subdirectory=publish/micropython-v1_24_1-esp32-esp32_generic_s3-stubs" --target typings
```

Josverl, 2025-02-05:
Fixed by : https://github.com/Josverl/micropython-stubs/releases/tag/v1.24.1-stdlib

`pip install micropython-stdlib-stubs==1.24.1`

closed 2025-02-05 by Josverl