Josverl/micropython-stubs #781: allow mypy to use custom stdlib stubs

Josverl, 2025-01-30:
in a Q&D test it is possible to allow mypy to use the 
new 1.24.1 stubs ( that omit sys) 

`mypy --custom-typeshed-dir ./typings .`
by adding : 
- `typings\mypy_extensions.pyi` 

origin : typestub repo : `stubs\mypy-extensions\mypy_extensions.pyi`

refs:
https://mypy.readthedocs.io/en/stable/command_line.html#cmdoption-mypy-custom-typeshed-dir

https://discuss.python.org/t/pep-561-s-module-resolution-order-seems-incorrect/55048

Josverl, 2025-02-05:
partially impemented in : 2959e982b44964e7efe95f314cce85be5c9e59e6

andrewleech, 2025-02-10:
Hi @Josverl following on from https://github.com/Josverl/micropython-stubs/issues/720 I've done a quick test of the v1.24.1 on my existing project with "patched/hacked stubs for mypu use" but quickly run into blockers that I think are related directly to me using the stubs `typing` dir as the custom_typing_dir for mypy
```
tools/typings/_mpy_shed/__init__.pyi:14: error: Invalid "type: ignore" comment  [syntax]
/home/.cache/pre-commit/repo_em78hv0/py_env-python3.10/lib/python3.10/site-packages/mypy/typeshed/stdlib/io.pyi:95: error: INTERNAL ERROR -- Please try using mypy master on GitHub:
https://mypy.readthedocs.io/en/stable/common_issues.html#using-a-development-mypy-build
If this issue continues with mypy master, please report a bug at https://github.com/python/mypy/issues
version: 1.7.0
/home/.cache/pre-commit/repo_em78hv0/py_env-python3.10/lib/python3.10/site-packages/mypy/typeshed/stdlib/io.pyi:95: : note: please use --show-traceback to print a traceback when reporting a bug
```

---
I did a _terrible_ job of tracking my changes when I was previously modding my installed stubs typings folder to make it work with mypy, such that now every time I get back to it to attempt to reverse engineer it into clean commits & PR here I don't get very far. Much of the problem is due to me inadvertently allowing vscode and/or black to auto-reformat all the files... so now diff's are very noisy.

I've still got an internally tracked task to clean this up properly and submit it, but little foreseable time to get to it so in liew of that I'm going to dump a snapshot here, just in case it helps you at all. Feel free to ignore it if it's more effort than its worth.

[typings_andrewleech_tgz.zip](https://github.com/user-attachments/files/18726658/typings_andrewleech_tgz.zip)
Note this file is supposed to be a `.tar.gz` but github wasn't allowing it so I've added a fake `.zip` to the end of the name that should be removed.

For context I believe this was originally installed with: 
```
pip install -U micropython-stm32-stubs --target ./tools/typings --no-user
```
and appears to be from `micropython_stm32_stubs==1.19.1.post9` & `micropython_stdlib_stubs==0.9.0`
Then black or ruff have reformatted it and I've made changes since.

I use pre-commit with `.pre-commit-config.yaml` and  `pyproject.toml` attached here, again with a fake `.zip` appended to the name.

[.pre-commit-config.yaml.zip](https://github.com/user-attachments/files/18726671/default.pre-commit-config.yaml.zip)

[pyproject.toml.zip](https://github.com/user-attachments/files/18726672/pyproject.toml.zip)

Additionally, this was another WIP typings folder I had, also for use as a typeshed dir, but trying to resolve https://github.com/Josverl/micropython-stubs/issues/720 in a different way.
[typings_mod_tgz.zip](https://github.com/user-attachments/files/18726700/typings_mod_tgz.zip)
While micropython implements streams as a "protocol" this concept doesn't really map to cpython. It's similar to a base class though, so I added a Stream base class to `stdlib/io.pyi` and used it as a baseclass for things like `io.StringIO`. To be honest though I don't remember how close this was to working correctly.

Josverl, 2025-02-10:
Thanks for sharing.

I run all all typecheck tests with the [same config](https://github.com/Josverl/micropython-stubs/blob/main/tests/quality_tests/_configs/pyproject.toml) 
```toml
exclude = [
    "typings[\\/].*", # TOML basic string 
]
```
I explicitly exclude the `typings` folder itself for both pyright and mypy, takinh a hint from typeshed , which also racks up a lot of errors it you turn a typechecker towards it.

wrt to the micropython io , 
I am still trying to better understand where the differences are, and how to express them well enough.
to allow multiple modules to use common micropython (base) classes I have moved these to a `_mpy_shed` module , that I now include in micropython-stdlib-stubs.
The Micropython io base classes are in [`_mpy_shed/io_mp.pyi`](https://github.com/Josverl/micropython-stubs/blob/main/micropython-reference/_mpy_shed/io_mp.pyi) but they will need more work.
but as some of the io methods are special-cased in the type checkers , it is hard/impossible to just alter.

Adding the `typings\mypy_extensions.pyi` allows mypy to 'accept' changes to the stdlib modules, but i cant add it by default as that then fails other tests.
so that will need to be an extras to allow mypy users to opt in to stdlib

Josverl, 2025-02-10:
If you want to submit changes , then the best angle is to target `micropython-reference`, 
as all type info flows from there to the stub packages.

Josverl, 2025-02-10:
## V1.19.1.post9  
compared with : 
`pip install micropython-stdlib-stubs==0.9.0  micropython-stm32-stubs==1.19.1.post9 --target typings --no-user ` 

Added / changed /: 
 - [ ] asyncio/*.pyi       - needed to fix asyncio stubs 
 - [ ] uasyncio/*.pyi      - needed to fix asyncio stubs
 - [ ] remove machine.PWM class  
 - [ ] change pyb.Timer.channel method signature
 - [ ] change time.time from method to class 

## V1.21.0 (.mod)
compared with: 
`pip install micropython-stdlib-stubs==1.0.0  micropython-stm32-stubs==1.21.0.post1 --target typings --no-user ` 

note: the re-install of stdlib is slightly different - while it does have the same version, perhaps changes were copied over from an earlier version 

Added / changed /: 
 - [ ] asyncio/*.pyi       - needed to fix asyncio stubs 
 - [ ] del VERSIONS.dis file
 - [ ] stdlib/os 
        - from . import path as _path - commented out
        - update statvfs_result ( minor change )
 - [ ] stdlib/path.pyi - renamed/removed
 - [ ] stdlib/io.Stream class added 
 - [ ] io 
    - add `# type: ignore [no-redef]` to a lot of methods
    - changes base class from IO to Stream 
 - [ ] machine.IDLE addedd
 - [ ] machine.ADC class constants added
 - [ ] uctypes.struct : add `__getattr__` method
    `def __getattr__(self, a): ...`
 - [ ] _typeshed - slightly newer version 
 - [ ] _codecs - slightly newer version
 - [ ] contextlib.pyi - slightly newer version
 - [ ] typing / typing_extensions - slightly newer version
 - [ ] os - slightly newer version

these are the diffs , ruff format on both sides to get rid of some noise, and leaving out the async patches and some metadata 

[changes_v1.19.1.patch](https://github.com/user-attachments/files/18742133/changes_v1.19.1.patch)
[changes_v1.20.0.patch](https://github.com/user-attachments/files/18742136/changes_v1.20.0.patch)

If you can still remember why you made which changes, that would help me decide what and how to integrate these in the stub project.

Josverl, 2025-03-01:
io.Stream is a Protocol in MicroPython.

[// Stream protocol](https://github.com/micropython/micropython/blob/2c80d369988a84d85efe0483f9cf469ff27b70f9/py/stream.h#L69-L92)

so it probably should be typed as a Protocol
```py 
class Stream(Protocol):
    """
    MicroPython stream protocol. Due to implementation mechanism
    not all methods are guaranteed to be available on all classes
    based on the stream type / protocol.
    """
    def __init__(self, *argv, **kwargs) -> None: ...
    def __enter__(self: Self) -> Self: ...
    def __exit__(
        self,
        exc_type: type[BaseException] | None,
        exc_val: BaseException | None,
        exc_tb: TracebackType | None,
    ) -> None: ...
    def close(self) -> None: ...
    def flush(self) -> None: ...
    def read(self, __size: int | None = ...) -> bytes: ...
    def read1(self, __size: int = ...) -> bytes: ...
    def readinto(self, __buffer: WriteableBuffer) -> int: ...
    def readline(self, __size: int | None = ...) -> bytes: ...
    def readlines(self, __hint: int = ...) -> list[bytes]: ...
    def seek(self, __offset: int, __whence: int = ...) -> int: ...
    def tell(self) -> int: ...
    def write(self, __buffer: ReadableBuffer) -> int: ...
    def write1(self, __buffer: ReadableBuffer) -> int: ...
```

Josverl, 2025-03-03:
Solved the majority of the mypy false positives by: 

 - adding `typings\stubs\mypy-extensions\mypy_extensions.pyi` to the stdlib stubs 
 - removing the duplication of several std modules 
 - various changes to the stdlib and the reference stubs 
 - updates to the stubber merge command 
 - updates to the micropython-stdlib build script

Josverl, 2025-03-07:
@andrewleech , as far as I can test the stubs now work without any known issue with mypy.

Do you want to test them  before I release them ?
it would be good if you can , 
and if/when you run into issues, I'd appreciate if you can give me a sip of the code to reproduce , and fix remaining issues.

One thing that I have not yet incorporated is the `Stream`  class, as I could not find a CPython version of that.
It may make sense to add a number of MicroPython Specific Protocols, but for now im happy to have `mypy` in a much better state.

Let me know your thoughts

andrewleech, 2025-03-09:
Hi @Josverl thanks for the feedback and investigations, sorry I haven't managed to follow along! I've managed to get bogged down / stuck assisting on a project using a 15 year old 16bit renesas mcu... it makes me love micropyton all the more but hasn't left much time to tinker.

I'm happy to try some testing however I'm struggling to get it to run; how are you installing this current test setup?
Are you referring to changes in you https://github.com/Josverl/micropython-stubs/compare/mypy_v3 branch?
I haven't got set up to run your stubber myself to generate a typings dir from source yet.

Regardless if you're happy with the progress I suggest releasing without waiting for me, I can certainly test a release against a few of my application codebases and look for discrepancies. 
If I find troubling errors reported against valid (micropython) code I'm happy to provide a test case for it definitely!

To be honest the older app that I worked on my patched typings module previously is throwing a lot of errors against said patched module, I clearly left it in a poor state on that branch anyway so it's not the best reference to compare against.

The Stream stuff is an interesting one for me, yeah the stub class approach I was attempting was due to it not having a matching concept in cpython. I wasn't really aware of the Protocol pattern used there though, that makes sense to lean on. It's a tough one to get "right" though as not all micropython classes that implement Stream protocol provide all the functions, so they don't all actually fit that when used as a parent class. That is definitely a separate topic though, can be figured out in future for sure.

Josverl, 2025-03-10:
Sorry , should have included install instructions.

No need to install stubber, 
The stubs are already generated and stored in git ( perhaps they should be stored in releases , but this works..)  

So you just just need to install them to a folder using pip :

```bash
pip install "git+https://github.com/Josverl/micropython-stubs@mypy_v3#subdirectory=publish/micropython-stdlib-stubs" --target typings
pip install "git+https://github.com/Josverl/micropython-stubs@mypy_v3#subdirectory=publish/micropython-v1_24_1-stm32-stubs" --target typings --no-deps
```

The mypy configuration I use in testing is :
```toml
[tool.mypy]
platform = "linux"
mypy_path = "typings"
custom_typeshed_dir = "typings" # allow mypy to use micropython-stdlib in typings/stdlib 
files = "*.py"
exclude = [
    "typings[\\/].*", # TOML basic string 
]

follow_imports = "silent"
follow_imports_for_stubs = true
no_site_packages = true
check_untyped_defs = true

# the below are to ease qa testing 
# disable_error_code = ["no-redef", "assignment", "var-annotated"]

# snippets may re-use the same variable names
# allow_redefinition = true

# other optional settings
# warn_unused_ignores = true  # allows to verify the absense of methods/classes in stubs 
```

Ill do a bit more tinkering and then push them out later this week

closed 2025-03-14 by Josverl