Josverl/micropython-stubs #806: Instantiating a Generic class returns None

markusand, 2025-03-17:
When trying to instantiate a simple Generic class, it returns None. Removing the (Generic[T]) part returns the proper instance.

```python
from typing import Generic, TypeVar

T = TypeVar('T')

class TestClass(Generic[T]):
    _attr: T
    
    def __init__(self, attr: T):
        self._attr = attr

test = TestClass(34)
print(test)  # None
```

I'm using micropython 3.4.0; MicroPython v1.25.0-preview.365.g3823aeb0f.dirty on 2025-03-10.
Don't know how to get typing version, but should definitely be more than 1.24.0

Josverl, 2025-03-17:
Are you trying to use Generics at **runtime**, or just as static type checking ?

I Have not yet found a way to get Generics working in runtime MicroPython, as ( to my understanding) it lacks some of the 'plumming' that it relies on: 
- metaclasses do not yet exist in MicroPython  
- classes do not support `def __class_getitem__(cls, params)` 

See the [rt_typing repo](https://github.com/Josverl/rt_typing) for some of the experiments:

## versions

good point - for the stubs installed in a typing folder the version can be seen in the folder structue 
```sh
tree typings
Folder PATH listing for volume DEV
Volume serial number is E61F-E2CD
D:\MYPYTHON\!-STUBTESTPROJECTS\RT_TYPING\TYPINGS
├───asyncio
├───micropython_rp2_stubs-1.23.0.post1.dist-info
├───micropython_stdlib_stubs-1.1.2.dist-info
├───stdlib
```

but there is no version yet in the typing.py or typing_extensions.py.
Should be useful to add a __version__ attribute in both.

[ ] add a __version__ attribute to mip/typings.py and typing_extensions.py

markusand, 2025-03-17:
I'm not sure about the answer. Generic is used for type checking during development, but it is definitely included at runtime. However, the syntax for Generic is the same as for class inheritance, so (from my limited understanding) wouldn't extending Subscriptable be enough to do 'nothing' and just pass? 

I was wondering if this issue was more related to micropython itself instead of this repo.

Josverl, 2025-03-17:
Adding Generic, and the underlying support is something that calls for a discussing in the MicroPython repo.
As far as I can tell (see above) the minimum needed is that `def __class_getitem__(cls, params)` to be able to allow `Generic[foo]` as part of the language , not quite sure if metaClasses are needed as well.

Josverl, 2025-05-12:
Some possible options are dicussed in : https://github.com/orgs/micropython/discussions/9327,
Or need to look into adding `__class_getitem__` in micropython core.
https://zetcode.com/python/dunder-class_getitem/

```py 

class _Subscriptable:
    def __getitem__(self, sub):
        return object

Generic = GenericClass()

```