Josverl/micropython-stubs #814: Installing micropython-stubs from uv

PaszaVonPomiot, 2025-04-25:
I noticed the [documentation](https://micropython-stubs.readthedocs.io/en/main/11_install_stubs.html) recommends installing micropython-stubs using pip with the `--target` flag.

I'm using [uv](https://github.com/astral-sh/uv) as a package manager, but uv does not support the `--target` option.
Is there an alternative way you recommend for installing micropython-stubs with uv, or would you consider documenting a uv-compatible method?

Thanks!

Josverl, 2025-04-25:
I usually do that using pyproject.toml ,and using --target works fine for me 
I also use it in CI/CD , again without a hitch , faster and fewer errors than PIP for which i needed to add a complete caching layer to get a single run under 30 minutes ( per version , per architecture ) .

What command do you use , and what error do you get ?

```toml
[project]
name = "esp32_c3"
authors = [{ name = "Jos Verlinde", email = "jos_verlinde@hotmail.com"}]

description = "My MicroPython project"
version = "0.1.0"
classifiers = [
  "Programming Language :: Python :: 3",
  "Programming Language :: Python :: MicroPython",
  "License :: OSI Approved :: MIT License"
]
readme = "readme.md"
license = "MIT"

[project.optional-dependencies]
# install to folder typings
# uv pip install -r pyproject.toml --extra stubs --target typings 
stubs = [
  "micropython-esp32-stubs"
]

```

PaszaVonPomiot, 2025-04-27:
I think I got confused by VS Code MyPy extension but after testing I found that following works:
```
uv add micropython-rp2-stubs  # stubs go to .venv\lib\site-packages
```
or adding following lines to `pyproject.toml`
```toml
[project.optional-dependencies]
stubs = [
  "micropython-rp2-stubs"
]
```
and running
```
uv pip install -r pyproject.toml --extra stubs --target typings
```

Thanks for help!

closed 2025-04-27 by PaszaVonPomiot