Josverl/micropython-stubs #817: Issues with typing and @no_type_check

DavideRossi, 2025-05-01:
I just tried installing typing to my Pico.
Then I took the PIO example from https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_1hz.py and just added `from typing import no_type_check` and `@no_type_check` before the pio function.
When running I get `TypeError: object 'closure' isn't a tuple or list` on the first line after the pio function.
I'm pasting here the whole code.

```Python
# Example using PIO to blink an LED and raise an IRQ at 1Hz.

import time
from machine import Pin
import rp2
from typing import no_type_check

@no_type_check
@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def blink_1hz():
    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    irq(rel(0))
    set(pins, 1)
    set(x, 31)                  [5]
    label("delay_high")
    nop()                       [29]
    jmp(x_dec, "delay_high")

    # Cycles: 1 + 7 + 32 * (30 + 1) = 1000
    set(pins, 0)
    set(x, 31)                  [6]
    label("delay_low")
    nop()                       [29]
    jmp(x_dec, "delay_low")

# Create the StateMachine with the blink_1hz program, outputting on Pin(25).
sm = rp2.StateMachine(0, blink_1hz, freq=2000, set_base=Pin(25))

# Set the IRQ handler to print the millisecond timestamp.
sm.irq(lambda p: print(time.ticks_ms()))

# Start the StateMachine.
sm.active(1)

```

Josverl, 2025-05-01:
Thanks for reporting,

What versions of the rp2 and stdlib stubs are installed?
Please share other configuration details from pyproject or vscode configuration

DavideRossi, 2025-05-01:
The issue is not on the development machine side. If I upload the script and I run it from MicroPython's REPL I get the same error.
In any case I'm using `micropython-rp2-rpi_pico-stubs==1.25.0.post1` on `MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico with RP2040`. I'm also using VSCode's MicroPico extension, so I have little visibility of that it set up for me.
In any case I'm willing to provide more details, just please explain how I should gather the info you need.

Josverl, 2025-05-02:
Thanks for clarification. You also installed the runtime `typing.mpy` module to your board.
I can reproduce the error now.

the problem lies in the runtime definition of the `@no_type_check` decorator that I had modeled after a decorator.
That seems to cause a problem when decorating a function with multiple decorators. I have seen that sometimes in my testing, but attributed that to other factors.

```py 
# simplest
def no_type_check(func):
    return func

# caused problems
# def no_type_check(func):
#     def wrapper(*args, **kwargs):
#         func(*args, **kwargs)
#     return wrapper
```
With the fix your code runs.

Ill make an update. 
You will need to update the `typings.mpy` on your board(s) 

## PIO code 

As your example uses PIO code , you may want to consider using type-hinsts for your PIO code, instead of disabling type checs for that section.

```py 
# Example using PIO to blink an LED and raise an IRQ at 1Hz.

import time
from machine import Pin
import rp2

# -----------------------------------------------
# add type hints for the rp2.asm_pio PIO Instructions
TYPE_CHECKING = False
if TYPE_CHECKING:
    from rp2.asm_pio import *
# -----------------------------------------------

@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def blink_1hz():
    # Cycles: 1 + 1 + 6 + 32 * (30 + 1) = 1000
    irq(rel(0))
    set(pins, 1)
    set(x, 31)                  [5]
    label("delay_high")
    nop()                       [29]
    jmp(x_dec, "delay_high")

    # Cycles: 1 + 7 + 32 * (30 + 1) = 1000
    set(pins, 0)
    set(x, 31)                  [6]
    label("delay_low")
    nop()                       [29]
    jmp(x_dec, "delay_low")
```

for more information : https://micropython-stubs.readthedocs.io/en/main/32_rp2_asm.html

DavideRossi, 2025-05-02:
Yes I knew about the PIO type hints but having stumbled on the typing issue I reported it. I can confirm that it's fixed now.

Josverl, 2025-05-02:
Much appreciated.

closed 2025-05-02 by Josverl