Josverl/micropython-stubs #819: Stub for `asyncio` falls back to the (incompatible) `stdlib` version.

nickovs, 2025-05-18:
Currently no stub is produced for `asyncio` and the stub for `uasyncio` simply does `from asyncio import *`. As a result the `asyncio` stub from `stdlib` is loaded.

The standard library has a host of classes and functions that don't exist on MicroPython and it is also missing some of the functions that are MicroPython specific (such as `asyncio.sleep_ms()`). It would be great is a correct MicroPython `asycnio` stub could be included.

Josverl, 2025-05-19:
Do you have your type checkers properly configured to use the updated typeshed stubs that are part of Micropython-stdlib-stubs?

The [sleep_ms method](https://github.com/Josverl/micropython-stubs/blob/main/publish%2Fmicropython-stdlib-stubs%2Fhandcrafted%2Fasyncio%2F__init__.pyi#L20)is part of the stubs but apparently not found in your setup.

Note that asyncio has proven quite difficult to provide good stubs for, so I'm sure further improvements can be found

closed 2025-05-19 by nickovs

Josverl, 2025-05-19:
@nickovs , not a waste of time at all.
getting the configuration right is as important as having the correct stubs.
@Lukaskremla and I have collaborated on these things before.